ashaduri/gsmartcontrol #2: GSmartControl fails to start. Errors and warnings in terminal.

aritorhamo, 2021-03-26:
Hello!

When I try to start GSmasrtControl, it crashes. I've been using it successfully before, but the last time I used it was quite a long time ago, so I don't know when the problem started. I get to "Scanning system, please wait...", but then the window disappears. When I try to start the program from the terminal, I get the following:
.........................
<error> [gtk] _g_io_module_get_default: Found default implementation gvfs (GDaemonVfs) for ‘gio-vfs’
<error> [gtk] setenv()/putenv() are not thread-safe and should not be used after threads are created
<error> [gtk] posix_spawn avoided (workdir specified) (fd close requested) 
<error> [gtk] setenv()/putenv() are not thread-safe and should not be used after threads are created
<error> [gtk] setenv()/putenv() are not thread-safe and should not be used after threads are created
<error> [gtk] posix_spawn avoided (workdir specified) (fd close requested) 
<error> [gtk] setenv()/putenv() are not thread-safe and should not be used after threads are created
<warn>  [hz] Warning: exit: Command line did not parse.
<warn>  [app] execute_smartctl(): Smartctl binary did not execute cleanly.
<warn>  [app] StorageDevice::execute_device_smartctl(): Smartctl binary did not execute cleanly.
<error> [gtk] setenv()/putenv() are not thread-safe and should not be used after threads are created
<error> [gtk] posix_spawn avoided (workdir specified) (fd close requested) 
<error> [gtk] setenv()/putenv() are not thread-safe and should not be used after threads are created
<warn>  [hz] Warning: exit: Device open failed, or device did not return an IDENTIFY DEVICE structure.
<warn>  [app] SmartctlParser::parse_full(): No ATA sections could be parsed. Returning.
<error> [gtk] setenv()/putenv() are not thread-safe and should not be used after threads are created
<error> [gtk] posix_spawn avoided (workdir specified) (fd close requested) 
<error> [gtk] setenv()/putenv() are not thread-safe and should not be used after threads are created
<warn>  [hz] Warning: exit: Device open failed, or device did not return an IDENTIFY DEVICE structure.
<warn>  [app] SmartctlParser::parse_full(): No ATA sections could be parsed. Returning.
<error> [gtk] setenv()/putenv() are not thread-safe and should not be used after threads are created
<error> [gtk] posix_spawn avoided (workdir specified) (fd close requested) 
<error> [gtk] setenv()/putenv() are not thread-safe and should not be used after threads are created
<warn>  [hz] Warning: exit: Device open failed, or device did not return an IDENTIFY DEVICE structure.
<error> [gtk] setenv()/putenv() are not thread-safe and should not be used after threads are created
<error> [gtk] posix_spawn avoided (workdir specified) (fd close requested) 
<error> [gtk] setenv()/putenv() are not thread-safe and should not be used after threads are created
<warn>  [hz] Warning: exit: Device open failed, or device did not return an IDENTIFY DEVICE structure.
<warn>  [app] SmartctlParser::parse_full(): No ATA sections could be parsed. Returning.
<error> [gtk] setenv()/putenv() are not thread-safe and should not be used after threads are created
<error> [gtk] posix_spawn avoided (workdir specified) (fd close requested) 
<error> [gtk] setenv()/putenv() are not thread-safe and should not be used after threads are created
<warn>  [hz] Warning: exit: Device open failed, or device did not return an IDENTIFY DEVICE structure.
<warn>  [app] SmartctlParser::parse_full(): No ATA sections could be parsed. Returning.
<error> [gtk] setenv()/putenv() are not thread-safe and should not be used after threads are created
<error> [gtk] posix_spawn avoided (workdir specified) (fd close requested) 
<error> [gtk] setenv()/putenv() are not thread-safe and should not be used after threads are created
<warn>  [hz] Warning: exit: Device open failed, or device did not return an IDENTIFY DEVICE structure.
<warn>  [app] SmartctlParser::parse_full(): No ATA sections could be parsed. Returning.
Muistialueen ylitys (luotiin core-tiedosto)
.........................
The last line is something like "Memory area exceeded (core file was created)" in English.

I don't know if this is a problem with my Linux Mint 20.1 installation or GSmartControl, so I just decided to start asking here. My system seems to work without problems otherwise. Thanks!

ashaduri, 2021-04-12:
Hello,

Please try running gsmartcontrol with `--no-scan` option, something like `gsmartcontrol-root --no-scan`. If it doesn't crash, try manually adding your devices (`/dev/sda`, `/dev/sr0`, etc...) and see which one causes the crash.
When you pinpoint the problematic device (e.g. `/dev/sr0`), please create output.txt like this:
`sudo smartctl -x /dev/sr0 &> output.txt` and attach the file here.

Please also check if you have `adwaita-icon-theme` installed.

Thanks

aritorhamo, 2021-04-18:
Thanks for your reply, ashaduri. I'm sorry for the delay in my response.

When I do `gsmartcontrol-root --no-scan`, I get the following:
`<error> [gtk] _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
<error> [gtk] posix_spawn avoided (workdir specified) (fd close requested) 
<error> [gtk] setenv()/putenv() are not thread-safe and should not be used after threads are created`

P.S. Is it a good practice to use "insert code" formating, when quoting the output from the terminal?

ashaduri, 2021-04-18:
Did it crash when you ran it as `gsmartcontrol-root --no-scan`?

The "insert code" feature in the editor is for single lines only.

aritorhamo, 2021-04-18:
I'm sorry for not being more specific. No, the program didn't crash, despite the error messages shown in the terminal. When I try to add any device manually, the program crashes, with the following additional lines shown in the terminal:

<error> [gtk] setenv()/putenv() are not thread-safe and should not be used after threads are created
<error> [gtk] posix_spawn avoided (workdir specified) (fd close requested) 
<error> [gtk] setenv()/putenv() are not thread-safe and should not be used after threads are created
/usr/bin/gsmartcontrol-root: rivi 167:  6837 Muistialueen ylitys     (luotiin core-tiedosto) pkexec --disable-internal-agent /usr/sbin/gsmartcontrol "--no-scan"

The lines are the same for all devices, except for the four digits after "167:", which are different for each device.

Here's the Output.txt file you asked for:
[output.txt](https://github.com/ashaduri/gsmartcontrol/files/6332060/output.txt)

I have these two Adwaita icon themes installed (but not currently in use; I'm using Mint-Y):
adwaita-icon-theme 3.36.1-2ubuntu0.20.04.2
adwaita-icon-theme-full 3.36.1-2ubuntu0.20.04.2

P.S. When I copy error messages from the terminal and paste them here, the comment system for some reason strips away "<error>" from the beginning of the copied lines. The word is there when I compose the comment, but isn't shown on the preview nor the final comment. Perhaps it's normal, but I thought to mention about it.

ashaduri, 2021-06-21:
I was unable to reproduce this crash on Linux Mint 20.1 with SATA SSD. I tried MATE and Cinnamon desktops.

Can you please tell me which desktop you use?

Thanks

P.S. The error messages you're seeing are harmless, most of them should be hidden by default in the next version.

aritorhamo, 2021-06-21:
I'm sorry for not being more specific. My desktop is Cinnamon. GSmartControl seems to work now, though. I haven't ttried it between answering the latest questions here and this moment, so I don't know when the situation changed. It may have something to do with another problem that I had a few weeks after reporting the issue, namely that one of my two external drives stopped working properly, the symptoms being interrupted file transfers and frequent dismountings. The reason turned out to be a defective SATA cable, which I noticed by changing the cable between the two external drives The faulty cable (or more precisely the dismountings that it caused) probably messed up the drive book-keeping on the system a little. It still doesn't look quite right to me, but GSmartControl has started to work, although maybe not for that reason. I should mention that the faulty cable perhaps isn't the cause for the original problem, because I tested GSmartControl several times with the external drives detached before reporting. 

Anyway, it's good to have GSmartControl working again. The reason for the original issue may stay a mystery, but if you still have some questions, I'm glad to answer them.

ashaduri, 2021-06-21:
Thank you for clarifying.

I guess a routine update to one of the underlying libraries fixed the issue. At least it's good to know it's not hardware-specific.
Still, I've been getting this kind of crash reports on and off for a few years now, and I still can't get to the bottom of it.
I'll close this issue for now, but if it comes back again, please take note of the exact installed versions of gtk, gtkmm, cairo, and glibc.

Thanks!

closed 2021-06-21 by ashaduri

ashaduri, 2022-01-13:
A crash with similar symptoms has been fixed, see #13 .