ashaduri/gsmartcontrol #4: Support NVMe storage

ashaduri, 2021-04-23:
This is a feature request to track NVMe support.

To fully support NVMe, the following criteria have to be met:

1. Better-than-experimental [NVMe support in smartctl](https://www.smartmontools.org/wiki/NVMe_Support).
2. Full NVMe JSON output support in smartctl.
3. JSON input support in GSmartControl.
4. NVMe support in GSmartControl GUI, as NVMe information differs significantly from (S)ATA.

Any help is appreciated.

Etaash-mathamsetty, 2022-01-23:
yes please add nvme support I have to use a proprietary application otherwise

pnevmoslon, 2022-03-25:
There is https://github.com/linux-nvme/nvme-cli and it works well for my Samsung 960 EVO 250GB. Perhaps it is worth adding it as a backend to GSmartControl?

ashaduri, 2022-03-25:
> There is https://github.com/linux-nvme/nvme-cli and it works well for my Samsung 960 EVO 250GB. Perhaps it is worth adding it as a backend to GSmartControl?

Hi,
Is there any added benefit of using nvme-cli instead of smartctl?
It would be a lot easier to support smartctl's NVME as GSmartControl is transitioning to JSON support for (S)ATA as well.

pnevmoslon, 2022-03-25:
There is https://github.com/linux-nvme/nvme-cli and it works well for my Samsung 960 EVO 250GB. Perhaps it is worth adding it as a backend to GSmartControl? 

> 
Sorry, I had an old version of smartmontools installed and I was sure that it did not work with NVME. Of course, there is no point in adding nvme-cli, the latest version of smartmontools works well.

hamishmb, 2022-05-31:
Oh yeah, I reported this last year. IIRC the issue is that smartctl outputs the information differently for NVME disks for some reason.

[smartctl_output.txt](https://github.com/ashaduri/gsmartcontrol/files/8808764/smartctl_output.txt)

Attached is my nvme smartctl output for posterity.

ssnjimb, 2022-06-29:
I have a BC711 NVMe SK hynix 256GB (Dell 3511) and it shows as unknown model :-(

hamishmb, 2022-06-29:
In smartctl?

hamishmb, 2022-06-29:
Hmm, do you mean the GUI, GSmartControl?

The smartctl on my Linux Mint 20.3 (Ubuntu 20.04 LTS) install says it's version 7.1, so 1.1.4 would be very very old.

It is known that GSmartControl doesn't support NVME yet (hence the bug), but if this is a bug in smartctl it might be quite interesting. Run "smartctl -a <your_device>" to see what smartctl thinks.

1ucay, 2022-11-25:
Release 7.3 - https://github.com/smartmontools/smartmontools/releases

    C:\Program Files\GSmartControl>smartctl.exe -j --all /dev/sda
    {
      "json_format_version": [
        1,
        0
      ],
      "smartctl": {
        "version": [
          7,
          3
        ],
        "svn_revision": "5338",
        "platform_info": "x86_64-w64-mingw32-w11-b25247",
        "build_info": "(sf-7.3-1)",
        "argv": [
          "smartctl",
          "-j",
          "--all",
          "/dev/sda"
        ],
        "exit_status": 0
      },
      "local_time": {
        "time_t": 1669414621,
        "asctime": "Fri Nov 25 23:17:01 2022 SE"
      },
      "device": {
        "name": "/dev/sda",
        "info_name": "/dev/sda",
        "type": "nvme",
        "protocol": "NVMe"
      },
      "model_name": "Samsung SSD 970 PRO 512GB",
      "serial_number": "S463NF0MXXXXXXX",
      "firmware_version": "1B2QEXP7",
      "nvme_pci_vendor": {
        "id": 5197,
        "subsystem_id": 5197
      },
      "nvme_ieee_oui_identifier": 9528,
      "nvme_total_capacity": 512110190592,
      "nvme_unallocated_capacity": 0,
      "nvme_controller_id": 4,
      "nvme_version": {
        "string": "1.3",
        "value": 66304
      },
      "nvme_number_of_namespaces": 1,
      "nvme_namespaces": [
        {
          "id": 1,
          "size": {
            "blocks": 1000215216,
            "bytes": 512110190592
          },
          "capacity": {
            "blocks": 1000215216,
            "bytes": 512110190592
          },
          "utilization": {
            "blocks": 107557128,
            "bytes": 55069249536
          },
          "formatted_lba_size": 512,
          "eui64": {
            "oui": 9528,
            "ext_id": 384696446185
          }
        }
      ],
      "user_capacity": {
        "blocks": 1000215216,
        "bytes": 512110190592
      },
      "logical_block_size": 512,
      "smart_support": {
        "available": true,
        "enabled": true
      },
      "smart_status": {
        "passed": true,
        "nvme": {
          "value": 0
        }
      },
      "nvme_smart_health_information_log": {
        "critical_warning": 0,
        "temperature": 39,
        "available_spare": 100,
        "available_spare_threshold": 10,
        "percentage_used": 2,
        "data_units_read": 77788198,
        "data_units_written": 53882918,
        "host_reads": 1089040374,
        "host_writes": 751165311,
        "controller_busy_time": 3261,
        "power_cycles": 1502,
        "power_on_hours": 6373,
        "unsafe_shutdowns": 185,
        "media_errors": 0,
        "num_err_log_entries": 9712,
        "warning_temp_time": 0,
        "critical_comp_time": 0,
        "temperature_sensors": [
          39,
          43
        ]
      },
      "temperature": {
        "current": 39
      },
      "power_cycle_count": 1502,
      "power_on_time": {
        "hours": 6373
      }
    }

    C:\Program Files\GSmartControl>smartctl.exe -j --all /dev/sdb
    {
      "json_format_version": [
        1,
        0
      ],
      "smartctl": {
        "version": [
          7,
          3
        ],
        "svn_revision": "5338",
        "platform_info": "x86_64-w64-mingw32-w11-b25247",
        "build_info": "(sf-7.3-1)",
        "argv": [
          "smartctl",
          "-j",
          "--all",
          "/dev/sdb"
        ],
        "exit_status": 0
      },
      "local_time": {
        "time_t": 1669414327,
        "asctime": "Fri Nov 25 23:12:07 2022 SE"
      },
      "device": {
        "name": "/dev/sdb",
        "info_name": "/dev/sdb",
        "type": "nvme",
        "protocol": "NVMe"
      },
      "model_name": "KINGSTON SEDC1000M3840G",
      "serial_number": "50026B728XXXX",
      "firmware_version": "S68F0120",
      "nvme_pci_vendor": {
        "id": 9798,
        "subsystem_id": 9798
      },
      "nvme_ieee_oui_identifier": 9911,
      "nvme_total_capacity": 3840755982336,
      "nvme_unallocated_capacity": 0,
      "nvme_controller_id": 1,
      "nvme_version": {
        "string": "1.3",
        "value": 66304
      },
      "nvme_number_of_namespaces": 1,
      "nvme_namespaces": [
        {
          "id": 1,
          "size": {
            "blocks": 7501476528,
            "bytes": 3840755982336
          },
          "capacity": {
            "blocks": 7501476528,
            "bytes": 3840755982336
          },
          "utilization": {
            "blocks": 7501476528,
            "bytes": 3840755982336
          },
          "formatted_lba_size": 512,
          "eui64": {
            "oui": 9911,
            "ext_id": 172439069653
          }
        }
      ],
      "user_capacity": {
        "blocks": 7501476528,
        "bytes": 3840755982336
      },
      "logical_block_size": 512,
      "smart_support": {
        "available": true,
        "enabled": true
      },
      "smart_status": {
        "passed": true,
        "nvme": {
          "value": 0
        }
      },
      "nvme_smart_health_information_log": {
        "critical_warning": 0,
        "temperature": 31,
        "available_spare": 100,
        "available_spare_threshold": 10,
        "percentage_used": 0,
        "data_units_read": 67103406,
        "data_units_written": 60312101,
        "host_reads": 237642019,
        "host_writes": 246983288,
        "controller_busy_time": 29727,
        "power_cycles": 622,
        "power_on_hours": 4342,
        "unsafe_shutdowns": 56,
        "media_errors": 0,
        "num_err_log_entries": 0,
        "warning_temp_time": 0,
        "critical_comp_time": 0
      },
      "temperature": {
        "current": 31
      },
      "power_cycle_count": 622,
      "power_on_time": {
        "hours": 4342
      }
    }

HubKing, 2023-02-12:
On Windows, CrystalDiskInfo could show the information of my Samsung NVME like any other disks. Both GsmartControl and Gnome Disks could not read it. `nvme-cli` could show the same data as Crystal. Well, I could use the command, but that is not as convenient as using the GUI. If dealing with NVME is too much work, how about just install nvme-cli as a dependency and simply call it and parse its output?

Anyway, since all modern laptops and desktops now use NVME as the main, if not the only, storage, I think NVME support really is needed.

bjlockie, 2023-04-09:
My new laptop only has a nvme drive.

antonio-petricca, 2023-04-15:
Any chance to support this format?

bjlockie, 2023-04-15:
> Any chance to support this format?

Read the 1st post.
Support in smartctl is still experimental.
You're welcome to supply a patch if you don't want to wait.

antonio-petricca, 2023-04-17:
Hi , I will try doing it as soon as I will get enough time. :)

cfelicio, 2023-10-02:
> > Any chance to support this format?
> 
> Read the 1st post. Support in smartctl is still experimental. You're welcome to supply a patch if you don't want to wait.

looks like the latest smartctl (7.4) recently released has full support for NVME.

jccardot, 2024-03-09:
I've got a USB to NVME adapter with a RTL9210 chipset, supported by smartctl, but the drive shows as "unknown model" in gsmartcontrol.
Attaching verbose output of gsmartcontrol and 3 invocations of smartctl (the drive is well recognized and displays all the info in the CLI, even in json format).
I have technical and programmation knowledge and am willing to help.
[gsmartcontrol.txt](https://github.com/ashaduri/gsmartcontrol/files/14547593/gsmartcontrol.txt)
[smartctl.txt](https://github.com/ashaduri/gsmartcontrol/files/14547594/smartctl.txt)

jccardot, 2024-03-12:
I made a nvme version with statistics and error log based on antonio-petricca's version. Could it be worth considering?

DrMcCoy, 2024-03-27:
Some of mine:

[0_nvme0n1-i.json](https://github.com/ashaduri/gsmartcontrol/files/14772462/0_nvme0n1-i.json)
[0_nvme0n1-x.json](https://github.com/ashaduri/gsmartcontrol/files/14772465/0_nvme0n1-x.json)
[0_sda-i.json](https://github.com/ashaduri/gsmartcontrol/files/14772463/0_sda-i.json)
[0_sda-x.json](https://github.com/ashaduri/gsmartcontrol/files/14772464/0_sda-x.json)
[1_nvme0n1-i.json](https://github.com/ashaduri/gsmartcontrol/files/14772461/1_nvme0n1-i.json)
[1_nvme0n1-x.json](https://github.com/ashaduri/gsmartcontrol/files/14772460/1_nvme0n1-x.json)
[1_sda-i.json](https://github.com/ashaduri/gsmartcontrol/files/14772459/1_sda-i.json)
[1_sda-x.json](https://github.com/ashaduri/gsmartcontrol/files/14772458/1_sda-x.json)

jccardot, 2024-03-27:
here it is
SATA SSD:
[smartctl-a.txt](https://github.com/ashaduri/gsmartcontrol/files/14776786/smartctl-a.txt)
[smartctl-a2.txt](https://github.com/ashaduri/gsmartcontrol/files/14776784/smartctl-a2.txt)
SATA HD:
[smartctl-b.txt](https://github.com/ashaduri/gsmartcontrol/files/14776785/smartctl-b.txt)
[smartctl-b2.txt](https://github.com/ashaduri/gsmartcontrol/files/14776783/smartctl-b2.txt)
USB NVME:
[smartctl-c.txt](https://github.com/ashaduri/gsmartcontrol/files/14776787/smartctl-c.txt)
[smartctl-c2.txt](https://github.com/ashaduri/gsmartcontrol/files/14776781/smartctl-c2.txt)

TomaszGasior, 2024-03-28:
Here's output for my SSD disk. I hope, it helps. [smartctl.txt](https://github.com/ashaduri/gsmartcontrol/files/14792118/smartctl.txt)

scribblemaniac, 2024-04-07:
I'm looking forward to the NVME support :slightly_smiling_face:  Here are some more JSON outputs to help:
NVME SSDs: [1-x](https://github.com/ashaduri/gsmartcontrol/files/14896534/nvme_1-x.txt)/[1-i](https://github.com/ashaduri/gsmartcontrol/files/14896539/nvme_1-i.txt) *, [2-x](https://github.com/ashaduri/gsmartcontrol/files/14896562/nvme_2-x.txt)/[2-i](https://github.com/ashaduri/gsmartcontrol/files/14896564/nvme_2-i.txt) *, [3-x](https://github.com/ashaduri/gsmartcontrol/files/14896565/nvme_3-x.txt)/[3-i](https://github.com/ashaduri/gsmartcontrol/files/14896570/nvme_3-i.txt) *
SATA SSDs: [1-x](https://github.com/ashaduri/gsmartcontrol/files/14896686/satassd_1-x.txt)/[1-i](https://github.com/ashaduri/gsmartcontrol/files/14896687/satassd_1-i.txt) *, [2-x](https://github.com/ashaduri/gsmartcontrol/files/14896688/satassd_2-x.txt)/[2-i](https://github.com/ashaduri/gsmartcontrol/files/14896690/satassd_2-i.txt), [3-x](https://github.com/ashaduri/gsmartcontrol/files/14896692/satassd_3-x.txt)/[3-i](https://github.com/ashaduri/gsmartcontrol/files/14896694/satassd_3-i.txt)
SATA HDDs: [1-x](https://github.com/ashaduri/gsmartcontrol/files/14896699/satahdd_1-x.txt)/[1-i](https://github.com/ashaduri/gsmartcontrol/files/14896705/satahdd_1-i.txt) *, [2-x](https://github.com/ashaduri/gsmartcontrol/files/14896702/satahdd_2-x.txt)/[2-i](https://github.com/ashaduri/gsmartcontrol/files/14896706/satahdd_2-i.txt) *
USB Flash Drives: [1-x](https://github.com/ashaduri/gsmartcontrol/files/14896584/usbflash_1-x.txt)/[1-i](https://github.com/ashaduri/gsmartcontrol/files/14896585/usbflash_1-i.txt) †, [2-x](https://github.com/ashaduri/gsmartcontrol/files/14896589/usbflash_2-x.txt)/[2-i](https://github.com/ashaduri/gsmartcontrol/files/14896590/usbflash_2-i.txt) †, [3-x](https://github.com/ashaduri/gsmartcontrol/files/14896593/usbflash_3-x.txt)/[3-i](https://github.com/ashaduri/gsmartcontrol/files/14896594/usbflash_3-i.txt) †, [4-x](https://github.com/ashaduri/gsmartcontrol/files/14896597/usbflash_4-x.txt)/[4-i](https://github.com/ashaduri/gsmartcontrol/files/14896598/usbflash_4-i.txt) †
CD/DVD Drive: [1-x](https://github.com/ashaduri/gsmartcontrol/files/14896810/disk_1-x.txt)/[1-i](https://github.com/ashaduri/gsmartcontrol/files/14896812/disk_1-i.txt)
MicroSD (via USB adapter): [1-x](https://github.com/ashaduri/gsmartcontrol/files/14896609/microsd_1-x.txt)/[1-i](https://github.com/ashaduri/gsmartcontrol/files/14896612/microsd_1-i.txt) †
USB 3.25" Floppy Drive: [1-x](https://github.com/ashaduri/gsmartcontrol/files/14896645/floppy_1-x.txt)/[1-i](https://github.com/ashaduri/gsmartcontrol/files/14896646/floppy_1-i.txt) †
SATA HDDs (via USB adapter): [1-x](https://github.com/ashaduri/gsmartcontrol/files/14896658/satausb_1-x.txt)/[1-i](https://github.com/ashaduri/gsmartcontrol/files/14896659/satausb_1-i.txt), [2-x](https://github.com/ashaduri/gsmartcontrol/files/14896665/satausb_2-x.txt)/[2-i](https://github.com/ashaduri/gsmartcontrol/files/14896667/satausb_2-i.txt), [3-x](https://github.com/ashaduri/gsmartcontrol/files/14896669/satausb_3-x.txt)/[3-i](https://github.com/ashaduri/gsmartcontrol/files/14896674/satausb_3-i.txt), [4-x](https://github.com/ashaduri/gsmartcontrol/files/14896677/satausb_4-x.txt)/[4-i](https://github.com/ashaduri/gsmartcontrol/files/14896679/satausb_4-i.txt) (failed drive), [5-x](https://github.com/ashaduri/gsmartcontrol/files/14896680/satausb_5-x.txt)/[5-i](https://github.com/ashaduri/gsmartcontrol/files/14896682/satausb_5-i.txt)
BMC Virtual Drives: [1-x](https://github.com/ashaduri/gsmartcontrol/files/14896716/bmc_1-x.txt)/[1-i](https://github.com/ashaduri/gsmartcontrol/files/14896722/bmc_1-i.txt) (floppy) †, [2-x](https://github.com/ashaduri/gsmartcontrol/files/14896725/bmc_2-x.txt)/[2-i](https://github.com/ashaduri/gsmartcontrol/files/14896727/bmc_2-i.txt) (HDD) †

\* Serial number, WWN, and EUI64 have been redacted in these outputs
† Required adding the `-d scsi` argument to the command. Without this, the output looks like this: [x](https://github.com/ashaduri/gsmartcontrol/files/14896625/error-x.txt)/[i](https://github.com/ashaduri/gsmartcontrol/files/14896627/error-i.txt)

cryobry, 2024-04-17:
[My NVMe json.](https://pastebin.com/csM4cmrj)

ashaduri, 2024-05-14:
Thanks everyone for providing the outputs!

GSmartControl now fully supports NVMe drives (yay!), including self-tests.
There are minor things missing still (like attribute warnings), but I feel confident I can close this issue. Please open new issues if you find major problems.

Smartmontools 7.4 is required. Please note that NVMe support is still marked "experimental" in it.

Meanwhile, please help me test the "main" branch. There have been major changes in it (including defaulting to JSON parser for all drives).

closed 2024-05-14 by ashaduri

andreymal, 2024-05-14:
> help me test the "main" branch

A bit off-topic here, but I can't build it in my Arch Linux installation (tried GCC 13.2.1 and 14.1.1), it fails with the same error as GitHub Actions https://github.com/ashaduri/gsmartcontrol/actions/runs/9079753013/job/24949631989#step:5:38

ashaduri, 2024-05-14:
> A bit off-topic here, but I can't build it in my Arch Linux installation (tried GCC 13.2.1 and 14.1.1), it fails with the same error as GitHub Actions https://github.com/ashaduri/gsmartcontrol/actions/runs/9079753013/job/24949631989#step:5:38

Thanks, should be fixed now.

ashaduri, 2024-07-05:
@akvadrako 
You are correct.
1. It looks like self-tests are not supported on WD NVMe, either by the drive firmware or smartctl. I have the same experience with my own WD NVMes.
2. Self-tests are supported and seem to work on my Samsung 970 EVO NVMe. The output contains a `nvme_self_test_log` json key for such drives.
3. I need to hide the "Perform tests" context menu for unsupported drives. Thanks for noticing this.

akvadrako, 2024-07-05:
I see that it works in smartmontools 7.4 when you omit the namespace. https://www.smartmontools.org/ticket/1827

ashaduri, 2024-07-06:
> I see that it works in smartmontools 7.4 when you omit the namespace. https://www.smartmontools.org/ticket/1827

Thanks for finding this. It looks like it should be fixed in smartmontools 7.5, so I don't see the need to implement a workaround in the gsmartcontrol code (yet).

ashaduri, 2024-10-31:
Since smartmontools 7.4 may be widely deployed, a workaround for WD drives lacking self-tests has been implemented in the main branch.

ashaduri, 2024-11-07:
GSmartControl 2.0.0 has been released with NVMe support.
Testing is highly welcome and encouraged!
For the best results, please use smartmontools 7.4 or newer (Windows package already includes it).

Chealer, 2024-11-19:
I am pleased to report that upgrading from 1.1.3 to 2.0.0 finally brought reasonable support for my Intel SSD Pro 7600p Series (2018), on Windows 11 using the builtin smartmontools (7.4). Thank you very much to @ashaduri for this accomplishment.

The new version lets me open details and perform self-tests. I can perform a successful ~1 minute short self-test, which―perhaps due to my age―I find not that reassuring. Unfortunately, the "Extended Self-Test" also reports success, but is even shorter (less than 5 seconds) and not logged, which I interpret as evidence that it is in fact broken.

Output can be shown, but trying "Save As" reproducibly crashes GSmartControl.
[SSDPEKKF256G8L.json](https://github.com/user-attachments/files/17809612/SSDPEKKF256G8L.json)

ashaduri, 2024-11-19:
> I am pleased to report that upgrading from 1.1.3 to 2.0.0 finally brought reasonable support for my Intel SSD Pro 7600p Series (2018), on Windows 11 using the builtin smartmontools (7.4). Thank you very much to @ashaduri for this accomplishment.

Thank you for testing it!

> The new version lets me open details and perform self-tests. I can perform a successful ~1 minute short self-test, which―perhaps due to my age―I find not that reassuring. Unfortunately, the [...]

I have encountered weird situations with self-tests myself. For example, after running a short test, the drive would report it as "not running", and there would be no entry in the self-test log. After a few minutes, self-tests would start working again.
This looks like either a drive firmware issue, or a smartctl one. Smartctl 7.4 is still experimental in this regard, so I'm hoping that it will get fixed eventually.

> Output can be shown, but trying "Save As" reproducibly crashes GSmartControl. [SSDPEKKF256G8L.json](https://github.com/user-attachments/files/17809612/SSDPEKKF256G8L.json)

I was able to reproduce this, but have not found a solution yet. Filed as #67 .

Thanks!

luckusmiSU, 2024-11-30:
Hello, you can include support ADATA Legend (2TB)?
My json file:
```
{
  "json_format_version": [
    1,
    0
  ],
  "smartctl": {
    "version": [
      7,
      4
    ],
    "pre_release": false,
    "svn_revision": "5530",
    "platform_info": "x86_64-linux-6.8.0-49-generic",
    "build_info": "(local build)",
    "argv": [
      "smartctl",
      "-j",
      "--all",
      "/dev/nvme0n1"
    ],
    "messages": [
      {
        "string": "Read Self-test Log failed: Invalid Field in Command (0x002)",
        "severity": "error"
      }
    ],
    "exit_status": 4
  },
  "local_time": {
    "time_t": 1732985710,
    "asctime": "Sat Nov 30 19:55:10 2024 MSK"
  },
  "device": {
    "name": "/dev/nvme0n1",
    "info_name": "/dev/nvme0n1",
    "type": "nvme",
    "protocol": "NVMe"
  },
  "model_name": "ADATA LEGEND 850",
  "serial_number": "2O04291Q4D2F",
  "firmware_version": "V1223C0",
  "nvme_pci_vendor": {
    "id": 7361,
    "subsystem_id": 7361
  },
  "nvme_ieee_oui_identifier": 7371800,
  "nvme_total_capacity": 2048408248320,
  "nvme_unallocated_capacity": 0,
  "nvme_controller_id": 1,
  "nvme_version": {
    "string": "1.4",
    "value": 66560
  },
  "nvme_number_of_namespaces": 1,
  "nvme_namespaces": [
    {
      "id": 1,
      "size": {
        "blocks": 4000797360,
        "bytes": 2048408248320
      },
      "capacity": {
        "blocks": 4000797360,
        "bytes": 2048408248320
      },
      "utilization": {
        "blocks": 4000797360,
        "bytes": 2048408248320
      },
      "formatted_lba_size": 512,
      "eui64": {
        "oui": 7371800,
        "ext_id": 17867820335
      }
    }
  ],
  "user_capacity": {
    "blocks": 4000797360,
    "bytes": 2048408248320
  },
  "logical_block_size": 512,
  "smart_support": {
    "available": true,
    "enabled": true
  },
  "smart_status": {
    "passed": true,
    "nvme": {
      "value": 0
    }
  },
  "nvme_smart_health_information_log": {
    "critical_warning": 0,
    "temperature": 41,
    "available_spare": 100,
    "available_spare_threshold": 10,
    "percentage_used": 0,
    "data_units_read": 2893745,
    "data_units_written": 3308170,
    "host_reads": 41013329,
    "host_writes": 27490055,
    "controller_busy_time": 1340,
    "power_cycles": 48,
    "power_on_hours": 145,
    "unsafe_shutdowns": 14,
    "media_errors": 0,
    "num_err_log_entries": 0,
    "warning_temp_time": 0,
    "critical_comp_time": 0,
    "temperature_sensors": [
      41,
      46
    ]
  },
  "temperature": {
    "current": 41
  },
  "power_cycle_count": 48,
  "power_on_time": {
    "hours": 145
  },
  "nvme_error_information_log": {
    "size": 256,
    "read": 16,
    "unread": 0
  }
}
```

ashaduri, 2024-11-30:
> Hello, you can include support ADATA Legend (2TB)? My json file:

Can you indicate what exactly is not working?
I was able to load the JSON file as virtual device successfully.

luckusmiSU, 2024-11-30:
[image]
All

luckusmiSU, 2024-11-30:
```
I was able to load the JSON file as virtual device successfully.
```
interesting.

ashaduri, 2024-11-30:
> ![image](https://private-user-images.githubusercontent.com/92354102/391257067-fbf0a8d6-8639-42c2-afbe-41037e1ea562.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

Which version is this?

luckusmiSU, 2024-11-30:
1.1.4
```
ubuntu repo
```

ashaduri, 2024-11-30:
> 1.1.4 ubuntu repo

"GSmartControl 2.0.0 has been released with NVMe support."

luckusmiSU, 2024-11-30:
> "GSmartControl 2.0.0 has been released with NVMe support."

[image]
I built your program , it didn't run.

elsimnki, 2025-02-18:
Hello, I installed your latest exe for windows 11, the only drive is Unknown Model

elsimnki, 2025-02-18:
`{
  "json_format_version": [
    1,
    0
  ],
  "smartctl": {
    "version": [
      7,
      4
    ],
    "pre_release": false,
    "svn_revision": "5530",
    "platform_info": "x86_64-w64-mingw32-w11-b26100",
    "build_info": "(sf-7.4-1)",
    "argv": [
      "smartctl",
      "--info",
      "--health",
      "--capabilities",
      "--json=o",
      "pd0"
    ],
    "output": [
      "smartctl 7.4 2023-08-01 r5530 [x86_64-w64-mingw32-w11-b26100] (sf-7.4-1)",
      "Copyright (C) 2002-23, Bruce Allen, Christian Franke, www.smartmontools.org",
      "",
      "Read Device Identity failed: Input/output error",
      "",
      "If this is a USB connected device, look at the various --device=TYPE variants",
      "A mandatory SMART command failed: exiting. To continue, add one or more '-T permissive' options."
    ],
    "exit_status": 2
  },
  "local_time": {
    "time_t": 1739916122,
    "asctime": "Tue Feb 18 23:02:02 2025 "
  },
  "device": {
    "name": "pd0",
    "info_name": "pd0",
    "type": "ata",
    "protocol": "ATA"
  }
}`