ashaduri/gsmartcontrol #7: Cannot open Help menu links in gsmartcontrol-root

JohnVeness, 2021-07-02:
Hi, many thanks for this app. I am using 1.1.3 on Ubuntu Mate 21.04.

I notice that all links in the Help menu (Help, Support, About/Website and the mailto: links in About/Credits) do not work when using `gsmartcontrol-root` (they work in non-root `gsmartcontrol`).

Running `gsmartcontrol-root -v` I see:
```
<info>  [app] GscMainWindow::on_action_activated(): Action activated: "action_online_documentation"
Running Firefox as root in a regular user's session is not supported.  ($XAUTHORITY is /home/john/.Xauthority which is owned by john.)
<info>  [app] GscMainWindow::on_action_activated(): Action activated: "action_support"
Running Firefox as root in a regular user's session is not supported.  ($XAUTHORITY is /home/john/.Xauthority which is owned by john.)
<info>  [app] GscMainWindow::on_action_activated(): Action activated: "action_about"
Running Firefox as root in a regular user's session is not supported.  ($XAUTHORITY is /home/john/.Xauthority which is owned by john.)
Running Thunderbird as root in a regular user's session is not supported.  ($XAUTHORITY is /home/john/.Xauthority which is owned by john.)
```

ashaduri, 2021-07-13:
Hello,
Thanks for the bug report!

GSmartControl uses `gtk_show_uri_on_window()` to open URLs, so it does not have any say in which browser is used to open the URL.
Since GSmartControl itself is launched with root privileges, and Firefox doesn't support running in this mode, I guess the only workaround would be to check whether the launch was successful. If it was not, a dialog with the URL (for the user to copy) would be shown.
I will have to check whether an unsuccessful launch can be detected at all. If not, I'm not sure what else I could do (any help is appreciated!)

JohnVeness, 2021-07-13:
I can see why it's not a good idea to run a browser as root! I know very little about this, but maybe you can (fork a process and?) drop root at runtime before calling gtk_show_uri etc.?

hamishmb, 2022-09-13:
It would be good for security in general to launch the GUI as a normal user, then use pkexec to run smartctl when required, with a rule in /usr/share/polkit-1/actions to display an authentication dialog to the user with a nice message about what the privileges are needed for.

This would help on the more modern systems that don't allow X11 windows to run as root, and possibly avoids issues with Wayland in the future as well: https://wiki.archlinux.org/title/Running_GUI_applications_as_root#Wayland

ashaduri, 2022-09-13:
> It would be good for security in general to launch the GUI as a normal user, then use pkexec to run smartctl when required, with a rule in /usr/share/polkit-1/actions to display an authentication dialog to the user with a nice message about what the privileges are needed for.

While it's a good idea in principle, last time I checked polkit did not allow you to specify the "keep" timeout after gaining the rights. There is only mention of a "brief" period in the specification, which is suggested to be 5 minutes. Asking for a password every 5 minutes while the program is running would be bad.
Also, gsmartcontrol itself needs elevated access to files sometimes, so it's not only for "smartctl" or "twcli" or similar backend commands.

The "proper" solution would be to have a privileged subprocess do everything related to the system and communicate back and forth with the GUI. Implementing this (especially in a cross-platform way) is a lot of work though.

hamishmb, 2022-09-14:
That's a fair point. I wonder how that could be done in a cross-platform way.

On Linux I guess DBUS could be used, or maybe a socket, but I'm not sure how this might work on other platforms.