ashaduri/gsmartcontrol #13: GSmartControl crashes when it adds a device or scans manually

firepainting, 2021-08-14:
Hi,

This is similar to #2.

`gsmartcontrol-root --no-scan` does not crash, but after I `Ctrl R` or add a device `/dev/sda`, it crashes.

Here's my [output.txt](https://github.com/ashaduri/gsmartcontrol/files/6986465/output.txt)

I'm using:

- Linux Mint 20.2 Cinnamon
- Adwaita but ONLY for controls (not icons)

Used to run okay but "suddenly" crashes.

Thank you. I hope these are helpful.

Stack trace (and then log from the terminal, below it):

```
===================================================================
 Info                                                              
===================================================================
           PID: 17900 (gsmartcontrol)
           UID: 1000 (username)
           GID: 1000 (username)
        Signal: 11 (SEGV)
     Timestamp: Sat 2021-08-14 19:14:53 (1h 9min ago)
  Command Line: gsmartcontrol
    Executable: /usr/sbin/gsmartcontrol
 Control Group: /user.slice/user-1000.slice/user@1000.service/apps.slice/apps-org.gnome.Terminal.slice/vte-spawn-c6515cd1-72a6-4c1b-aca5-e9e774f02df7.scope
          Unit: user@1000.service
     User Unit: vte-spawn-c6515cd1-72a6-4c1b-aca5-e9e774f02df7.scope
         Slice: user-1000.slice
     Owner UID: 1000 (username)
       Boot ID: e5b6c0796ca34bf0b04e166ee7289bb1
    Machine ID: 939972095cf1459c8b22cc608eff85da
      Hostname: Hostname
       Storage: /var/lib/systemd/coredump/core.gsmartcontrol.1000.e5b6c0796ca34bf0b04e166ee7289bb1.17900.1628939693000000000000.lz4
       Message: Process 17900 (gsmartcontrol) of user 1000 dumped core.
                
                Stack trace of thread 17900:
                #0  0x0000561fe5e7911e n/a (gsmartcontrol + 0xe211e)
                #1  0x00007fc8972cccf5 n/a (libgtkmm-3.0.so.1 + 0x329cf5)
                #2  0x00007fc8967c804b n/a (libgtk-3.so.0 + 0x14704b)
                #3  0x00007fc8962db1b8 g_hash_table_foreach (libglib-2.0.so.0 + 0x401b8)
                #4  0x00007fc8967c7edf n/a (libgtk-3.so.0 + 0x146edf)
                #5  0x00007fc8967cd8bf n/a (libgtk-3.so.0 + 0x14c8bf)
                #6  0x00007fc896a6f815 n/a (libgtk-3.so.0 + 0x3ee815)
                #7  0x00007fc8963d8a56 n/a (libgobject-2.0.so.0 + 0x14a56)
                #8  0x00007fc8963f7b48 g_signal_emit_valist (libgobject-2.0.so.0 + 0x33b48)
                #9  0x00007fc8963f80f3 g_signal_emit (libgobject-2.0.so.0 + 0x340f3)
                #10 0x00007fc8967c9a73 gtk_cell_area_apply_attributes (libgtk-3.so.0 + 0x148a73)
                #11 0x00007fc8968a1a79 n/a (libgtk-3.so.0 + 0x220a79)
                #12 0x00007fc896723506 n/a (libgtk-3.so.0 + 0xa2506)
                #13 0x00007fc895907fea n/a (libatk-bridge-2.0.so.0 + 0x14fea)
                #14 0x00007fc8963eca8d n/a (libgobject-2.0.so.0 + 0x28a8d)
                #15 0x00007fc8963f7bbe g_signal_emit_valist (libgobject-2.0.so.0 + 0x33bbe)
                #16 0x00007fc8963f8b9c g_signal_emit_by_name (libgobject-2.0.so.0 + 0x34b9c)
                #17 0x00007fc8963d8802 g_closure_invoke (libgobject-2.0.so.0 + 0x14802)
                #18 0x00007fc8963ec243 n/a (libgobject-2.0.so.0 + 0x28243)
                #19 0x00007fc8963f7bbe g_signal_emit_valist (libgobject-2.0.so.0 + 0x33bbe)
                #20 0x00007fc8963f80f3 g_signal_emit (libgobject-2.0.so.0 + 0x340f3)
                #21 0x00007fc8968cbc98 gtk_list_store_insert (libgtk-3.so.0 + 0x24ac98)
                #22 0x00007fc897317cce _ZN3Gtk9ListStore6appendEv (libgtkmm-3.0.so.1 + 0x374cce)
                #23 0x0000561fe5e73be0 n/a (gsmartcontrol + 0xdcbe0)
                #24 0x0000561fe5e74c1f n/a (gsmartcontrol + 0xddc1f)
                #25 0x0000561fe5e76257 n/a (gsmartcontrol + 0xdf257)
                #26 0x0000561fe5e5d14a n/a (gsmartcontrol + 0xc614a)
                #27 0x0000561fe5e5667c n/a (gsmartcontrol + 0xbf67c)
[... 221 lines cut ...]
#22 0x00007fc897317cce in Gtk::ListStore::append() () from /lib/x86_64-linux-gnu/libgtkmm-3.0.so.1
#23 0x0000561fe5e73be0 in ?? ()
#24 0x0000561fe5e74c1f in ?? ()
#25 0x0000561fe5e76257 in ?? ()
#26 0x0000561fe5e5d14a in ?? ()
#27 0x0000561fe5e5667c in ?? ()
#28 0x0000561fe5e05fc9 in ?? ()
#29 0x00007fc895d540b3 in __libc_start_main (main=0x561fe5e05fa0, argc=1, argv=0x7fffa24f8b48, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffa24f8b38) at ../csu/libc-start.c:308
#30 0x0000561fe5e170ee in ?? ()
```
**Log from terminal**

```
<error> [gtk] _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
<error> [gtk] posix_spawn avoided (workdir specified) (fd close requested) 
<error> [gtk] setenv()/putenv() are not thread-safe and should not be used after threads are created
<error> [gtk] setenv()/putenv() are not thread-safe and should not be used after threads are created
<error> [gtk] posix_spawn avoided (workdir specified) (fd close requested) 
<error> [gtk] setenv()/putenv() are not thread-safe and should not be used after threads are created
<error> [gtk] setenv()/putenv() are not thread-safe and should not be used after threads are created
<error> [gtk] posix_spawn avoided (workdir specified) (fd close requested) 
<error> [gtk] setenv()/putenv() are not thread-safe and should not be used after threads are created
<warn>  [app] SmartctlParser::parse_full(): No ATA sections could be parsed. Returning.
/usr/bin/gsmartcontrol-root: line 167: 51916 Segmentation fault      (core dumped) pkexec --disable-internal-agent /usr/sbin/gsmartcontrol "--no-scan"
```

ashaduri, 2021-08-14:
Hi,
Thank you for the bug report and stack trace information.
Unfortunately, without debug symbols, it doesn't contain that much useful information.

Could you please install the following packages:
- `gsmartcontrol-dbgsym`
- `libgtk-3-0-dbgsym`
- `libgtkmm-3.0-1v5-dbgsym`
- `libglib2.0-0-dbgsym`
and retrieve the stack trace again?
You can remove these packages after we find the source of this bug.

Also, if a `core` file is generated during crash, please attach it here. I don't know how Mint handles core files, but I suspect, if `apport` service is enabled, `/var/crash` will contain the respective `core` file for the crash.

Thanks!

firepainting, 2021-08-15:
Hi,

Thank you for your response but I'm really sorry I can't help you further. I'm not that savvy a user and have no experience with this enterprise. 

I found the packages you stated but I find it a bit scary to install them as they seem to depend on quite a lot of other packages and could break others. 

I would like to get into this but I cannot really afford it right now, especially if something else gets broken. Meanwhile I might just look for another solution. Thanks for understanding!

Regarding the core file, I tested this again and found a file generated in `/var/lib/systemd/coredump` that looks something like `core.gsmartcontrol.0.76.[alpha-numeric string].72546.[numeric string].lz4` Is this what you're looking for?

ashaduri, 2021-08-15:
That's OK, no worries. Thanks anyway.

> I found the packages you stated but I find it a bit scary to install them as they seem to depend on quite a lot of other packages and could break others.

They probably depend on other `-dbgsym` packages. These packages simply install the "debug information files" that help gdb and have no effect on the system. However, I do understand that you don't want to risk damage to your system.

> core.gsmartcontrol.0.76.[alpha-numeric string].72546.[numeric string].lz4
Yes, most likely this is the "core" file that may help me pinpoint the crash location. I would appreciate it if you attached it to this bug report.

Thanks

ashaduri, 2022-01-12:
Thanks everyone, I was able to reproduce this with Linux Mint 20.3 Mate in Virtualbox.

Fixed in main branch: 3e4cefb04fc9e597c81b9d9997ddebf0e178d0e4.
Fixed in gsmartcontrol-gtk3 branch: 939914a5ab31cad3e73ad8105b46245bd7abe88f.

commit 939914a5 (2022-01-12): Fixed a crash when trying to render main window icons which have incomplete data; fixes #13 in gsmartcontrol-gtk3 branch.
commit 3e4cefb0 (2022-01-12): Fixed a crash when trying to render main window icons which have incomplete data; fixes #13 in main branch.
closed 2022-01-12 by ashaduri