ashaduri/gsmartcontrol #18: SCSI support

1ucay, 2021-11-07:
Please, can you add support for SCSI?

On my system I have 16 disc ( 14 SATA, 2 SAS )

PS C:\Program Files\GSmartControl> .\smartctl.exe -d scsi --all /dev/sdb
smartctl 6.6 2017-11-05 r4594 [x86_64-w64-mingw32-w10-b22494] (sf-6.6-1)
Copyright (C) 2002-17, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Vendor:               WDC
Product:              WUH721818AL5204
Revision:             C120
Compliance:           SPC-5
User Capacity:        18 000 207 937 536 bytes [18,0 TB]
Logical block size:   512 bytes
Physical block size:  4096 bytes
LU is fully provisioned
Rotation Rate:        7200 rpm
Form Factor:          3.5 inches
Logical Unit id:      0x5000cca28XXXXXX
Serial number:        3WXXXXXX
Device type:          disk
Transport protocol:   SAS (SPL-3)
Local Time is:        Sun Nov 07 16:25:09 2021 SE
SMART support is:     Available - device has SMART capability.
SMART support is:     Enabled
Temperature Warning:  Enabled

=== START OF READ SMART DATA SECTION ===
SMART Health Status: OK

Current Drive Temperature:     47 C
Drive Trip Temperature:        85 C

Manufactured in week 24 of year 2020
Specified cycle count over device lifetime:  50000
Accumulated start-stop cycles:  603
Specified load-unload count over device lifetime:  600000
Accumulated load-unload cycles:  647
Elements in grown defect list: 0

Vendor (Seagate) cache information
  Blocks sent to initiator = 1846517556051968

Error counter log:
           Errors Corrected by           Total   Correction     Gigabytes    Total
               ECC          rereads/    errors   algorithm      processed    uncorrected
           fast | delayed   rewrites  corrected  invocations   [10^9 bytes]  errors
read:          0       52         0        52     658912      17939,937           0
write:         0        1         0         1       4799      19434,312           0
verify:        0       39         0        39      12332          0,000           0

Non-medium error count:        0

No self-tests have been logged

ashaduri, 2021-11-08:
Hi,
It is highly unlikely that GSmartControl will support SCSI.
The SCSI output of smartctl is very different from (S)ATA. At most, when smartctl fully supports JSON output for SCSI, GSmartControl can display that information in the GUI, which is marginally better than displaying the text above.

1ucay, 2021-11-08:
I download smartctl.exe from 
https://github.com/smartmontools/smartmontools/releases

PS C:\Program Files\smartmontools> smartctl.exe -j --all /dev/sdb
{
  "json_format_version": [
    1,
    0
  ],
  "smartctl": {
    "version": [
      7,
      2
    ],
    "svn_revision": "5155",
    "platform_info": "x86_64-w64-mingw32-w10-b22494",
    "build_info": "(sf-7.2-1)",
    "argv": [
      "smartctl",
      "-j",
      "--all",
      "/dev/sdb"
    ],
    "exit_status": 2
  },
  "device": {
    "name": "/dev/sdb",
    "info_name": "/dev/sdb",
    "type": "ata",
    "protocol": "ATA"
  }
}

which lead to "exit_status": 2
so, you can try invoke new call with "-d scsi"

PS C:\Program Files\smartmontools> smartctl.exe -j -d scsi --all /dev/sdb
{
 "json_format_version": [
    1,
    0
  ],
  "smartctl": {
    "version": [
      7,
      2
    ],
    "svn_revision": "5155",
    "platform_info": "x86_64-w64-mingw32-w10-b22494",
    "build_info": "(sf-7.2-1)",
    "argv": [
      "smartctl",
      "-j",
      "-d",
      "scsi",
      "--all",
      "/dev/sdb"
    ],
    "exit_status": 0
  },
  "device": {
    "name": "/dev/sdb",
    "info_name": "/dev/sdb",
    "type": "scsi",
    "protocol": "SCSI"
  },
  "vendor": "WDC",
  "product": "WUH721818AL5204",
  "model_name": "WDC WUH721818AL5204",
  "revision": "C120",
  "scsi_version": "SPC-5",
  "user_capacity": {
    "blocks": 35156656128,
    "bytes": 18000207937536
  },
  "logical_block_size": 512,
  "physical_block_size": 4096,
  "rotation_rate": 7200,
  "form_factor": {
    "scsi_value": 2,
    "name": "3.5 inches"
  },
  "serial_number": "3WXXXXX",
  "device_type": {
    "scsi_value": 0,
    "name": "disk"
  },
  "local_time": {
    "time_t": 1636369580,
    "asctime": "Mon Nov 08 12:06:20 2021 SE"
  },
  "smart_status": {
    "passed": true
  },
  "temperature": {
    "current": 46,
    "drive_trip": 85
  },
  "power_on_time": {
    "hours": 2543,
    "minutes": 8
  },
  "scsi_grown_defect_list": 0,
  "scsi_error_counter_log": {
    "read": {
      "errors_corrected_by_eccfast": 0,
      "errors_corrected_by_eccdelayed": 52,
      "errors_corrected_by_rereads_rewrites": 0,
      "total_errors_corrected": 52,
      "correction_algorithm_invocations": 658919,
      "gigabytes_processed": "17940,302",
      "total_uncorrected_errors": 0
    },
    "write": {
      "errors_corrected_by_eccfast": 0,
      "errors_corrected_by_eccdelayed": 1,
      "errors_corrected_by_rereads_rewrites": 0,
      "total_errors_corrected": 1,
      "correction_algorithm_invocations": 4810,
      "gigabytes_processed": "19434,351",
      "total_uncorrected_errors": 0
    },
    "verify": {
      "errors_corrected_by_eccfast": 0,
      "errors_corrected_by_eccdelayed": 39,
      "errors_corrected_by_rereads_rewrites": 0,
      "total_errors_corrected": 39,
      "correction_algorithm_invocations": 12340,
      "gigabytes_processed": "0,000",
      "total_uncorrected_errors": 0
    }
  }
}

For me is only importatnt smart_status and temperature. What do you think?

ashaduri, 2021-11-08:
> For me is only importatnt smart_status and temperature. What do you think?

If GSmartControl is going to support it, it should support SCSI properly. Showing only 2 values doesn't make sense - you don't need GUI for that, you can easily extract that information using a script of some kind.

1ucay, 2021-11-08:
Ok nevermind :)

ashaduri, 2024-11-01:
Meanwhile, with the new JSON support, GSmartControl is able to show basic information for SCSI drives.
No plans for additional SCSI support yet.

1ucay, 2024-11-03:
Thank you, sir!