ashaduri/gsmartcontrol #20: gsmartcontrol 1.1.3 windows 64 works via installer but fails when unzipped

TonyApuzzo, 2021-11-29:
When I run gsmartcontrol from an unzipped folder, it fails, however, if I run it from an installed version using the .exe installer, then it works.  Note that the problem appears to be more related to smartctl-nc.exe than to gsmartcontrol itself. `smartctl-nc.exe -V` displays nothing which is why the gsmartcontrol launcher is failing immediately.

Is there more to running gsmartcontrol via the unzipped archive than just unzipping and running the command?  I get the following behavior when running from an unzipped folder:

[image]

Command Line log:

```
PS C:\Users\tonya\Downloads\gsmartcontrol-1.1.3-win64> .\gsmartcontrol.exe
PS C:\Users\tonya\Downloads\gsmartcontrol-1.1.3-win64>
PS C:\Users\tonya\Downloads\gsmartcontrol-1.1.3-win64>
<dump>  [app]   locale: 1
<dump>  [app]   version: 0
<dump>  [app]   hide_tabs: 1
<dump>  [app]   scan: 1
<dump>  [app]   arg_add_virtual: [empty]
<dump>  [app]   arg_add_device: [empty]
<dump>  [app]   arg_gdk_scale: nan
<dump>  [app]   arg_gdk_dpi_scale: nan
<dump>  [app] LibDebug options:
<dump>  [app]   levels_enabled: 11111
<dump>  [app]   debug_colorize: false
<dump>  [app] {anonymous}::app_init_config(): Global config file: "gsmartcontrol.conf"
<dump>  [app] {anonymous}::app_init_config(): Local config file: "C:\Users\tonya\AppData\Roaming\gsmartcontrol\gsmartcontrol.conf"
<dump>  [app] {anonymous}::app_init_config(): Old local config file: "C:\Users\tonya\Documents\gsmartcontrol.conf"
<info>  [rmn] Loading under "/config" from file "C:\Users\tonya\AppData\Roaming\gsmartcontrol\gsmartcontrol.conf".
<dump>  [rconfig] /                    [3] /
config               [4] /config
gui                  [3] /config/gui
main_window          [3] /config/gui/main_window
default_size_w       [3] /config/gui/main_window/default_size_w 560
default_size_h       [3] /config/gui/main_window/default_size_h 779
default_pos_x        [3] /config/gui/main_window/default_pos_x 234
default_pos_y        [3] /config/gui/main_window/default_pos_y 234
default              [4] /default
<info>  [rconfig] Setting autosave config file to "C:\Users\tonya\AppData\Roaming\gsmartcontrol\gsmartcontrol.conf"
<info>  [rconfig] Starting config autosave with 180 sec. interval.
<info>  [app] Current C locale: C
<info>  [app] Current C++ locale: C

(GSMART~1.EXE:23308): Gtk-WARNING **: Theme parsing error: gtk-win32-base.css:1043:59: Junk at end of value for background-image
<dump>  [app] Current GTK theme: win32

(GSMART~1.EXE:23308): Gtk-WARNING **: Could not find the icon 'drive-harddisk'. The 'hicolor' theme
was not found either, perhaps you need to install it.
You can get a copy from:
        http://icon-theme.freedesktop.org/releases
<info>  [app] hz::data_file_find(): Data file "icon_hdd.png" found at ".".
<info>  [app] hz::data_file_find(): Data file "icon_cddvd.png" found at ".".
<info>  [app] get_smartctl_binary(): Smartmontools installation not found in "HKLM\SOFTWARE\Microsoft\Windows\CurrentVersion\Uninstall\smartmontools\InstallLocation".
<dump>  [default] ENTER: "Cmdex::execute()"
<info>  [app] Cmdex::execute(): Executing "'smartctl-nc.exe' -V".
<dump>  [default] EXIT:  "Cmdex::execute()"
<dump>  [default] ENTER: "Cmdex::stopped_cleanup()"
<dump>  [default] ENTER: "Cmdex::unset_stop_timeouts()"
<dump>  [default] EXIT:  "Cmdex::unset_stop_timeouts()"
<dump>  [default] EXIT:  "Cmdex::stopped_cleanup()"
<dump>  [default] ENTER: "Cmdex::stopped_cleanup()"
```

TonyApuzzo, 2021-12-09:
I noticed today that the problem seems to be invoking from command-line rather than using the NSIS versus zip.

Even when using the installer, it gsmartcontrol still fails to work correctly if you run it from powershell or cmd.

it's not the current directory but something about running from shell.  I get the exact same bad result even if launching from PowerShell:

```powershell
    PS C:\Users\tonya\scoop\apps\gsmartcontrol\1.1.3> start "C:\ProgramData\Microsoft\Windows\Start Menu\Programs\GSmartControl.lnk"
```

ashaduri, 2022-01-15:
Hi,

Thanks for the report.

This should be fixed in 0183626c734df336b8a09f6d7481e64c00de749d and 235fba8a04687e73149bedd15741342c530d0edc, at least it fixed the issue for me when CWD is different from location of `gsmartcontrol.exe`, and the algorithm no longer depends on PATH.

closed 2022-01-15 by ashaduri