ashaduri/gsmartcontrol #23: Wrong parsing 1TB capacity

MikePryadko, 2022-01-07:
### System
OS: `Xubuntu 20.04.3 LTS`
kernel: `Linux 5.11.0-43-generic #47~20.04.2-Ubuntu SMP Mon Dec 13 11:06:56 UTC 2021`
smartctl: `7.1 2019-12-30 r5022 [x86_64-linux-5.11.0-43-generic] (local build)`
gsmartcontrol: `1.1.3`
hdd: `Seagate/Samsung ST1000LM024 (HN-M101MBB)`

### Problem
`smartctl` (tab "**General**") says `1 TB`, but `gsmartcontrol` (tab "**General**" -> button "**View Output**" -> window "**Smartctl Output**") shows `1 B`

[image]

[image]

ashaduri, 2022-01-08:
Hello,
Thanks for reporting this.

I tried, but cannot reproduce the bug.

Can you please attach the smartctl output here? You can save it right from the "Smartctl Output" window.

Also, please attach the output of "locale" command.

Thanks!

MikePryadko, 2022-01-08:
```
$ locale 
LANG=en_US.UTF-8
LANGUAGE=en_US:en
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC=ru_RU.UTF-8
LC_TIME=ru_RU.UTF-8
LC_COLLATE="en_US.UTF-8"
LC_MONETARY=ru_RU.UTF-8
LC_MESSAGES="en_US.UTF-8"
LC_PAPER=ru_RU.UTF-8
LC_NAME=ru_RU.UTF-8
LC_ADDRESS=ru_RU.UTF-8
LC_TELEPHONE=ru_RU.UTF-8
LC_MEASUREMENT=ru_RU.UTF-8
LC_IDENTIFICATION=ru_RU.UTF-8
LC_ALL=
```
<details>
<summary>smartl output:</summary>

```
smartctl 7.1 2019-12-30 r5022 [x86_64-linux-5.11.0-43-generic] (local build)
Copyright (C) 2002-19, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Family:     Seagate Samsung SpinPoint M8 (AF)
Device Model:     ST1000LM024 HN-M101MBB
Serial Number:    S30YJ9JH601896
LU WWN Device Id: 5 0004cf 2122fa208
Firmware Version: 2BA30001
User Capacity:    1 000 204 886 016 bytes [1,00 TB]
Sector Sizes:     512 bytes logical, 4096 bytes physical
Rotation Rate:    5400 rpm
Form Factor:      2.5 inches
Device is:        In smartctl database [for details use: -P show]
ATA Version is:   ATA8-ACS T13/1699-D revision 6
SATA Version is:  SATA 3.0, 6.0 Gb/s (current: 1.5 Gb/s)
Local Time is:    Sat Jan  8 12:02:33 2022 MSK
SMART support is: Available - device has SMART capability.
SMART support is: Enabled
AAM feature is:   Disabled
APM level is:     128 (minimum power consumption without standby)
Rd look-ahead is: Enabled
Write cache is:   Enabled
DSN feature is:   Unavailable
ATA Security is:  Disabled, NOT FROZEN [SEC1]

=== START OF READ SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED

General SMART Values:
Offline data collection status:  (0x00)	Offline data collection activity
					was never started.
					Auto Offline Data Collection: Disabled.
Self-test execution status:      (   0)	The previous self-test routine completed
					without error or no self-test has ever 
					been run.
Total time to complete Offline 
data collection: 		(12300) seconds.
Offline data collection
capabilities: 			 (0x5b) SMART execute Offline immediate.
					Auto Offline data collection on/off support.
					Suspend Offline collection upon new
					command.
					Offline surface scan supported.
					Self-test supported.
					No Conveyance Self-test supported.
					Selective Self-test supported.
SMART capabilities:            (0x0003)	Saves SMART data before entering
[... 182 lines cut ...]
  77    2022-01-08 09:55    39  ********************
  78    2022-01-08 10:00    40  *********************
 ...    ..( 11 skipped).    ..  *********************
  90    2022-01-08 11:00    40  *********************
  91    2022-01-08 11:05    39  ********************
 ...    ..(  2 skipped).    ..  ********************
  94    2022-01-08 11:20    39  ********************
  95    2022-01-08 11:25    40  *********************
  96    2022-01-08 11:30    40  *********************
  97    2022-01-08 11:35    23  ****
  98    2022-01-08 11:40    27  ********
  99    2022-01-08 11:45    29  **********
 100    2022-01-08 11:50    31  ************
 101    2022-01-08 11:55    32  *************
 102    2022-01-08 12:00    32  *************

Write SCT (Get) Error Recovery Control Command failed: Read of ATA output registers not implemented [JMicron]
SCT (Get) Error Recovery Control command failed

Device Statistics (GP/SMART Log 0x04) not supported

ATA_READ_LOG_EXT (addr=0x11:0x00, page=0, n=1) failed: 48-bit ATA commands not implemented [JMicron]
Read SATA Phy Event Counters failed
```
</details>

ashaduri, 2022-01-10:
Thanks, I was able to reproduce the problem with 1.1.3.

This particular issue (when running smartctl or loading existing output) was originally fixed in the main branch in cb1abab75fa0687ae2389e5d90c305f70c9e8f8d.
A fix for all locales (when running smartctl) has been implemented in ad014517b9c1e45ecf01df312f07ec370acf12a8.
In the future, JSON output parser should let us avoid these kinds of issues altogether.

closed 2022-01-10 by ashaduri