ashaduri/gsmartcontrol #28: the amount of data recorded on the SSD in human-friendly form

OmlineEditor, 2022-02-25:
I ask you to make it possible to see the amount of recorded data on the SSD in a human-friendly form, where the data is in terabytes, and not in clusters or short ones still need to be interpreted correctly.

ashaduri, 2022-02-28:
Hi,
Please post a screenshot of which attributes or statistics you mean.
I think some SSDs already report this in GiB, so I'd like to know which values you're interested in.

OmlineEditor, 2022-02-28:
do something like this:
[image]

ashaduri, 2022-02-28:
Thanks.
Meanwhile, you can check the Attributes tab, and at least on my SSD, there is a "Total Written (GiB)" attribute, which should be easier to read.

OmlineEditor, 2022-02-28:
I don't have a value in my attributes for how much data is recorded

ferdnyc, 2022-04-25:
Just to +1 this a bit, on my TEAM SSD the only read/write attributes are "Total LBAs (Read|Written)" which are even more inscrutable. So, having the stats values translated into human-readable units seems handy.

hamishmb, 2022-09-13:
Most of mine are the same and don't present it nicely in an attribute, sadly.

ferdnyc, 2023-02-02:
@OmlineEditor 

> [image]

I did just want to (very belatedly) note one thing: The "Lifetime power-on resets" attribute is just a flat _count_ of the number of times the device has been... well... powered on. 

(Must be a laptop, as my desktop systems show double-digit values. My very oldest drives are low-triple-digits, but that's after 8+ years of runtime. :grin:)

Point is, it doesn't translate to "59 days" or any other value. Any attribute named "_Count" — which is actually most of them, in the majority of drives — generally can't be units-translated, as they just hold a unitless value: "number of times this particular event has been recorded".

Your SSD seems to be an exception to the "mostly `_Count` attributes" rule, in that the screenshot is showing a craaaazy number of temperature stats. (Wish mine were that informative!)

One _could_ argue, I suppose, that it would be nice to make temp values optionally translatable into Fahrenheit, for us dirty Americans, or into Kelvin, for exactly nobody. But since nearly all of those many temp values are just `0` (and not 0 Celsius), even that would be useless, at least for your drive.

Then there are other attributes, like (on spinning-rust HDDs) `Spin_Up_Time` and `Seek_Time_Performance`, that really give _**no indication at all**_ what units they're expressed in. 
* Are those even documented anywhere? (I assume... probably, if you know where to look?)
* Are the units/scale consistent between different vendors? (HAHAHA! ...Yeah, good luck with that.)
* `Spin_Up_Time` is _maaaybe_ in milliseconds, but who knows? (And even if it is, translating millisecond values to any other scale is... not really useful. `232 ms` => `0.232 s`? "Great, thanks.")

Ultimately, there are a **very** limited number of attributes for which any sort of human-readable value translation/reformatting could be done at all, and even fewer for which it could be done **reliably/usefully**. (Though I'm still in favor of doing so, for those few attributes.)

ashaduri, 2023-02-02:
> Your SSD seems to be an exception to the "mostly _Count attributes" rule, in that the screenshot is showing a craaaazy number of temperature stats. (Wish mine were that informative!)

Just as a note, some drives have these stats in the Temperature log (presented in an ASCII way in the Temperature Log tab).

> One _could_ argue, I suppose, that it would be nice to make temp values optionally translatable into Fahrenheit, for us dirty Americans, or into Kelvin, for exactly nobody. But since nearly all of [...]

One thing I can say for sure, I will not be implementing C->F translation ;)
In fact, I never noticed that computer temperatures were presented in anything other than Celsius - all the benchmarks I've seen always used C. Then again, maybe I just didn't pay enough attention as long as C was present.

> * Are those even documented anywhere? (I assume... probably, if you know where to look?)

Wikipedia (I know...) and random mailing lists were my source for most of the "help info" tooltips. The newer attributes and stats are less documented though.

> Ultimately, there are a **very** limited number of attributes for which any sort of human-readable value translation/reformatting could be done at all, and even fewer for which it could be done **reliably/usefully**. (Though I'm still in favor of doing so, for those few attributes.)

I think I'll just humanize some those few stats where it does make sense.
Luckily, smartctl mostly takes care of the rest of them in the Attributes tab.

ferdnyc, 2023-02-03:
> In fact, I never noticed that computer temperatures were presented in anything other than Celsius - all the benchmarks I've seen always used C. Then again, maybe I just didn't pay enough attention as long as C was present.

Honestly, I don't think they are. Even us Americans have gotten used to seeing things like CPU temps expressed in Celsius.

> One thing I can say for sure, I will not be implementing C->F translation ;)

Yay! :+1:

closed 2026-10-25 by ashaduri