ashaduri/gsmartcontrol #30: segmentation fault

Lanchon, 2022-03-25:
GSmartControl version 1.1.3
linux mint 19.3 (ubuntu bionic)
4.15.x kernel

gsmartcontrol shows a windows then immediately segfaults. this happens on my user; if i create a different user on the same machine i cannot repro there.

log and strace here:
https://gist.github.com/Lanchon/7e23edf11ec25bf66e41c4e63ff2f4ae

hope it helps!

ashaduri, 2022-03-26:
Thanks,
This is most likely #13, which has been fixed in version 1.1.4.
If you still get an error with 1.1.4, please reopen the issue.

closed 2022-03-26 by ashaduri

Lanchon, 2022-03-26:
> This is most likely https://github.com/ashaduri/gsmartcontrol/issues/13, which has been fixed in version 1.1.4.

@ashaduri yep, you are right. thank you!!!