ashaduri/gsmartcontrol #38: [Question] How to install v1.1.4 on Mint 19.3 ?

KaKi87, 2022-07-08:
**Version and Environment**
 - GSmartControl version: experiencing #13 with v1.1.3, trying to install v1.1.4
 - OS: Linux Mint 19.3 based on Ubuntu 18.04 (EOL: April 2023)

**Describe the Issue**
`apt policy gsmartcontrol` outputs :
```
gsmartcontrol:
  Installed: 1.1.3-1
  Candidate: 1.1.3-1
```
`apt install ./gsmartcontrol_1.1.4-1_amd64.deb` ([downloaded from ftp.de.debian.org](http://ftp.de.debian.org/debian/pool/main/g/gsmartcontrol/gsmartcontrol_1.1.4-1_amd64.deb)) outputs :
```
The following packages have unmet dependencies:
 gsmartcontrol : Depends: libatkmm-1.6-1v5 (>= 2.28.2) but 2.24.2-3 is to be installed
                 Depends: libc6 (>= 2.33) but 2.27-3ubuntu1.6 is to be installed
                 Depends: libgcc-s1 (>= 3.0) but it is not installable
                 Depends: libglibmm-2.4-1v5 (>= 2.66.2) but 2.56.0-1 is to be installed
                 Depends: libgtkmm-3.0-1v5 (>= 3.24.5) but 3.22.2-2 is to be installed
                 Depends: libpangomm-1.4-1v5 (>= 2.46.1) but 2.40.1-4 is to be installed
                 Depends: libstdc++6 (>= 11) but 8.4.0-1ubuntu1~18.04 is to be installed
```

---
Thanks

ashaduri, 2022-07-08:
GSmartControl's [Downloads](https://gsmartcontrol.shaduri.dev/downloads) page shows you how to do it:

https://software.opensuse.org//download.html?project=home%3Aalex_sh%3Agsmartcontrol%3Astable_latest&package=gsmartcontrol

KaKi87, 2022-07-08:
Perfect !

But, where was I supposed to find that link ?

The only one I found was : https://gsmartcontrol.shaduri.dev/downloads

Thanks

closed 2022-07-08 by KaKi87

ashaduri, 2022-07-08:
Quoting the page you just linked (see the second link in the text):

If the package contained in your favourite distribution or repository is not up-to-date yet, our own [OBS Project Directory](http://download.opensuse.org/repositories/home:/alex_sh:/gsmartcontrol:/stable_latest/) contains the latest GSmartControl packages for a number of Linux distributions. Please [select the distribution](https://software.opensuse.org//download.html?project=home%3Aalex_sh%3Agsmartcontrol%3Astable_latest&package=gsmartcontrol) to install these packages.

KaKi87, 2022-07-08:
My bad, thanks.