ashaduri/gsmartcontrol #39: Error log tab shown in red, but no errors shown

hamishmb, 2022-09-13:
**Version and Environment**
 - GSmartControl version: 1.1.3
 - OS: Linux Mint 20.3 Cinnamon x86_64

**Describe the Bug**
If errors in the SMART log exist but don't have any description, the tab is shown in red, but the table in the GUI is empty.

[image]

[image]

(you can see the errors noted in the text log below)

I don't know how often this happens in the real world, and I'd guess not often, but it would be good if the table had entries to show that the error log entry is empty/has no description.

**To Reproduce**

Not necessarily very easy to reproduce, as you need a drive that exhibits this weird behaviour.

Steps to reproduce the behavior:
1. Plug in a drive that has the issue.
2. Go to the error log tab and see the issue.

If there's another way I can help, let me know - I'm going to guess it's going to be pretty hard to reproduce this.

**Screenshots**
See above.

**Additional Context**
Kernel version: Linux 5.15.0-46-generic #49~20.04.1-Ubuntu SMP Thu Aug 4 19:15:44 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

Graphical environment: Cinnamon (X11)

GTK version: GTK3 3.24.20

hamishmb, 2022-09-13:
Sorry, didn't mean to assign you, I didn't realise it was going to do that.

ashaduri, 2022-09-14:
Hi,
This is probably a bug in smartctl output parser.
Can you please save the smartctl output (there's a "Save As" button at the bottom of that window) and attach it here?

hamishmb, 2022-09-14:
Sure, here it is.

[2-Power_SSD2042A_240GB_AA001470300567000035_2022-09-14.txt](https://github.com/ashaduri/gsmartcontrol/files/9564914/2-Power_SSD2042A_240GB_AA001470300567000035_2022-09-14.txt)

I had a quick look at the changelog and it didn't seem like anything related to this was fixed, but I'm happy to try it if you like/can't reproduce?

ashaduri, 2024-05-15:
The Smartctl output shows that the error log entries are empty, so it's not a parser bug.

This is either:
- a bug in smartctl;
- or (more likely) the drive firmware does not report the actual errors in the log, but reports only the error count. Could be a bug in firmware.

There's nothing GSmartControl can do here, unfortunately.

closed 2024-05-15 by ashaduri