ashaduri/gsmartcontrol #40: Build with pcre2

Vascom, 2022-12-06:
Could you plead add support for build gsmartcontrol with pcre2.
Old pcre is deprecated at Fedora Linux so I need rebuild with pcre2.

https://fedoraproject.org/wiki/PcreDeprecation
https://bugzilla.redhat.com/show_bug.cgi?id=2128306

ashaduri, 2022-12-07:
GSmartControl currently depends on pcrecpp, which is a part of pcre, but not pcre2.
There are two options to handle this:
- Port pcrecpp to pcre2 (not sure about this). This may be the easy way.
- Port GSmartControl to C++ regex. This will require some very good tests as C++ implementations vary quite a lot.

Most of the pcrecpp code in GSmartControl will also lose much of its usefulness once the JSON support is fully implemented.

Anyway, since Fedora does seem to be giving the developers plenty of time for migration, I will be continuing the main branch development and porting the non-JSON parser to C++. If a time comes that the pcre1 support is removed from Fedora and the C++ support is not done yet, I'll look into porting pcrecpp to pcre2.

Thanks
:ukraine:

skitt, 2023-03-19:
See also https://bugs.debian.org/1000074 (for Debian).

dvzrv, 2024-04-17:
Hi! :wave: 

On Arch Linux we have also started the process to replace pcre with pcre2 as the former is now unmaintained:
https://archlinux.org/todo/move-to-pcre2/

Removing the need for pcre would therefore be much appreciated :pray:

ashaduri, 2024-05-16:
The program has been ported to std::regex (needs testing).
The main branch no longer depends on PCRE.

closed 2024-05-16 by ashaduri