ashaduri/gsmartcontrol #42: Update Drive Database Error

boxerfan88, 2023-03-18:
**Version and Environment**
 - GSmartControl version: 1.1.4
 - OS: Windows 10

**Describe the Bug**
When trying to execute Update Drive Database, it will throw an error "Failed to execute child process (Bad file descriptor)".
I can successfully execute update-smart-drivedb.exe from the program folder.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Options'
2. Click on 'Update Drive Database'
3. See error

**Screenshots**

[image]

**Additional Context**

HubKing, 2023-07-11:
There is no "Discussion" tab, so I have to ask here.  On Linux, it update fails if `xterm` does not exist. Since I use another, better terminal app, I had uninstalled xterm. Is the xterm an important part of the Linux, like VI, that all systems are supposed to have? Otherwise, can't GSmartControl use any available terminal app or ... I don't know, does it really need a terminal app to update the database?

ashaduri, 2023-07-15:
> There is no "Discussion" tab, so I have to ask here. On Linux, it update fails if `xterm` does not exist. Since I use another, better terminal app, I had uninstalled xterm. Is the xterm an important [...]

Yes, it needs xterm. If you installed a GSmartControl package, it should have included xterm as its dependency. If not, please file a bug report against that package. The xterm package on my distribution is 38K, so you won't gain much by uninstalling it.
The reason that only xterm is supported is two-fold:
- Some terminal apps have different command invocation syntax.
- The default terminal app selection mechanism is desktop-specific, while GSmartControl aims to be very cross-platform.

On the other hand, wayland-based desktops are getting stable, so another mechanism will have to be implemented soon-ish.

ashaduri, 2024-11-05:
Fixed on Windows in git main branch.

closed 2024-11-05 by ashaduri