ashaduri/gsmartcontrol #46: Pre-releases of smartctl no longer work due to "pre-" prefix in version info

chrfranke, 2023-06-20:
**Version and Environment**
 - GSmartControl version: 1.1.4
 - OS: Windows 10 22H2 (but affects all platforms)

**Describe the Bug**
Pre-releases of smartctl no longer work due to "pre-" prefix in version info

**To Reproduce**
Steps to reproduce the behavior:
1. Install a CI release from https://builds.smartmontools.org/ or build it from source.
2. Select "Look for smartctl in smartmontools installation directory first" in GSmartControl preferences.
3. Press ^R
4. "Unknown model" is shown for all drives.

**Additional Context**
First smartctl output line of CI builds after smartmontools 7.3:
`smartctl pre-7.4 2023-06-13 r5481 [x86_64-w64-mingw32-w10-22H2] (CircleCI)`

If the "pre-" is removed, GSmartControl works again. So current GSmartControl 1.1.4 will work with upcoming smartmontools 7.4 release.

ashaduri, 2023-06-21:
Hi Christian,

Thanks for letting me know. I will fix this in the main branch.
I'm not sure it's critical enough to be backported to 1.1 branch though.

Thank you!

chrfranke, 2023-06-21:
You're welcome. Backporting is IMO not needed as there is possibly no benefit for GSmartControl's use cases with a 7.4 pre-release instead of a 7.3 release and with the upcoming 7.4 release it will work again.

chrfranke, 2023-08-01:
FYI: New official release 7.4 works with GSmartControl 1.1.4, so no need to backport anything.

OT: In next GSmartControl release, you might want to show the drive database version number instead of `Yes/No`. The related smartctl line is now for example:
```
Device is:        In smartctl database 7.3/5528
Device is:        Not in smartctl database 7.3/5528
```

ashaduri, 2023-08-01:
Christian, thanks for the suggestion, I should display the version there.
Hopefully, I will have enough time to finally finish the JSON port as well.

ashaduri, 2024-05-16:
This has now been implemented in the main branch.

Thanks!

ashaduri, 2024-11-04:
> OT: In next GSmartControl release, you might want to show the drive database version number instead of `Yes/No`. The related smartctl line is now for example:
> 
> ```
> Device is:        In smartctl database 7.3/5528
> Device is:        Not in smartctl database 7.3/5528
> ```

Something similar to this has now been implemented. Thanks again for the suggestion!