ashaduri/gsmartcontrol #47: Self test execution bar color should change to RED when error are present (currently remains green) and title should state "test completed with errors"

ohadschn, 2023-07-08:
**Version and Environment**
 - GSmartControl version: 1.1.4 (latest released)
 - OS: Windows 10 22H2

**Is your feature request related to a problem? Please describe.**
The color green is universally interpreted as "successful". I run self tests regularly and recently one of them failed. However I almost missed it as everything seemed normal at first glance.
1. The title was "test completed"
2. The execution bar was green

It was only upon further inspection that I realized that the subtitle mentioned the test "completed with read failure".

**Describe the solution you'd like**
Upon any test failure:
1. The title should change to "test completed with errors"
2. The execution bar color should change to RED

**Additional context**
[image]

ashaduri, 2023-07-15:
Hi,

Unfortunately, the progress bar color is determined by the theme and it would be problematic to change it.

However, I agree that the failure should be more "in your face". Actually, there is a "warning" image displayed, but due to a bug in Windows package, it's missing.

So, fixing this bug requires:
- Fixing the missing image in Windows package.
- Maybe changing the color of the text to red to make it stand out.

ohadschn, 2023-07-15:
What about changing the test "test completed" to something like "TEST FAILED" in bold and/or red?

ashaduri, 2023-07-15:
> What about changing the test "test completed" to something like "TEST FAILED" in bold and/or red?

Yes, this is also a good idea.

ashaduri, 2024-11-04:
This issue has been fixed in the main branch - the test results are presented in a much more obvious manner now.

closed 2024-11-04 by ashaduri