ashaduri/gsmartcontrol #56: Tests stall when switching desktops in Gnome

Tom-H-L, 2023-12-29:
**Version and Environment**
 - GSmartControl version: 1.1.4
 - OS: Ubuntu 22.10

**Describe the Bug**
When running disk tests on one desktop of Gnome and then switching to another desktop for doing other works, when returning back to the desktop where GSmartControl runs, the test of GSmartControl crashes/stalls and the entire app and harddisk need to be restarted.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect a USB hard disk (Winchester disk, I have not tried it with an SDD) and power it on.
2. Launch GSmartControl on desktop 1 and start a disk test, e.g. Short offline
3. Switch to desktop 2 and do some other work for some seconds, e.g. browse the internet. In the background you can hear the HDD continue to do the tests
4. Switch back to desktop 1 before the test completes
5. Click on the test window
6. In 1 out of 3-4 such desktop switches, you notice that the test crashes/ stalls. Suddenly no more disk test sounds can be heard and no more progress in GSmartControlthe can be seen. The test window cannot be closed because "Close" is grayed out. 
7. Close the GSmartControl main window so that all windows including the test window close. Then power off the HDD, too, else it won't be accessible by GSmartControl after its restart. Then restart both GSmartControl and the HDD
8. Redo everything and notice that it is reproducible
9. If the desktop is NOT switched, but always focuses on GSmartControl doing the test, the test completes at 100% without any problems.
10. The problem arises only, when the desktop is switched after the launch of the test and during the running test you return back to the dekstop of GSmartControl. When you just switch to another desktop and not return to the desktop of GSmartControl while the test is running, the test completes without problems. So the problem is with switching desktop back to GSmartControl during a running test.

ashaduri, 2024-11-19:
Hi Tom,
Sorry for getting to you so late.

There have been some changes in the way GSmartControl handles windows.
Do you still get this issue with 2.0.0?

I would say the issue sounds very strange. The fact that the drive becomes inoperable is really weird - perhaps, when polling for progress, the smartctl process hangs and the drive becomes confused.

Thanks

Tom-H-L, 2024-11-19:
Hi Alexander! 

I am still on 1.1.4. I will retest as soon as 2.x will be available in the Debian repos and I have upgraded.

P.S.: Happy to see that this screenshot still lives up to today in the docs: https://gsmartcontrol.shaduri.dev/screenshots/3ware-raid.png 🤩