ashaduri/gsmartcontrol #58: smartd integration

ohadschn, 2024-01-06:
**Version and Environment**
 - GSmartControl version: latest released (1.1.4)
 - OS: Windows 10 22H2

**Is your feature request related to a problem? Please describe.**
I like to monitor my hard drives on a periodic basis. Currently, I need to manually check the SMART status and run the self-tests every week, and compare them to the results from last week (which I have to remember for each of my drives).

**Describe the solution you'd like**
It appears `smartmontools` already has a solution for this, namely `smartd`. It would be great to have it integrated into `gsmartcontrol` where I could configure it, monitor its status, etc.

**Describe alternatives you've considered**
Configure and run `smartd` without a GUI, and periodically verify that it's running properly (`smartd status`).

**Additional context**
Even though the `d` in `smartd` stands for "daemon" which is a linux term, it seems to be implemented for Windows as well: https://sourceforge.net/projects/smartmontools/

ashaduri, 2024-01-10:
I'm sorry to say that monitoring is out of scope of this project. GSmartControl is for examining SMART data and running tests only.

closed 2024-01-10 by ashaduri

ohadschn, 2024-01-11:
> I'm sorry to say that monitoring is out of scope of this project. GSmartControl is for examining SMART data and running tests only.

That's fair, I found this project which migh fit the bill: https://github.com/AnalogJ/scrutiny?tab=readme-ov-file