ashaduri/gsmartcontrol #65: suing latest smartctl (7.5) on the gui it's showing at startup "there was an error ..."

vasile2024, 2024-10-17:
Using latest GSMARTCONTROL e latest build of SMARTCTL.. when starting GsmartControl utility
is showing "There was an error while executing smartctl"
smartctl returning invalid output

in the show output there is:

```
smartctl pre-7.5 2024-10-16 r5625 [x86_64-w64-mingw32-win8.1] (CircleCI)
Copyright (C) 2002-24, Bruce Allen, Christian Franke, www.smartmontools.org

smartctl comes with ABSOLUTELY NO WARRANTY. This is free
software, and you are welcome to redistribute it under
the terms of the GNU General Public License; either
version 2, or (at your option) any later version.
See https://www.gnu.org for further details.

smartmontools pre-release 7.5
smartmontools SVN rev 5625 dated 2024-10-16 at 14:27:12
smartmontools build host: x86_64-w64-mingw32
smartmontools build with: C++11, GCC 12-win32, MinGW-w64 10.0.0
smartmontools configure arguments: [hidden in reproducible builds]
reproducible build SOURCE_DATE_EPOCH: 1729088833 (2024-10-16 16:27:13)

```

using it on WIndows 8.1 64

vasile2024, 2024-10-17:
selecting re-sacn i obtain "!UNKNWON" for any device connected... and pressing the infosd about a NVME drive i receive an error:

```
smartctl pre-7.5 2024-10-16 r5625 [x86_64-w64-mingw32-win8.1] (CircleCI)
Copyright (C) 2002-24, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Number:                       INTEL HBRPEKNX0101AH
Serial Number:                      BTTE92652EKD256D-1
Firmware Version:                   HPS1
PCI Vendor/Subsystem ID:            0x8086
IEEE OUI Identifier:                0x5cd2e4
Controller ID:                      1
NVMe Version:                       1.3
Number of Namespaces:               1
Namespace 1 Size/Capacity:          256.060.514.304 [256 GB]
Namespace 1 Formatted LBA Size:     512
Namespace 1 IEEE EUI-64:            5cd2e4 579190a384
Local Time is:                      Thu Oct 17 14:08:04 2024 
Firmware Updates (0x14):            2 Slots, no Reset required
Optional Admin Commands (0x0016):   Format Frmw_DL Self_Test
Optional NVM Commands (0x005f):     Comp Wr_Unc DS_Mngmt Wr_Zero Sav/Sel_Feat Timestmp
Log Page Attributes (0x0f):         S/H_per_NS Cmd_Eff_Lg Ext_Get_Lg Telmtry_Lg
Maximum Data Transfer Size:         32 Pages
Warning  Comp. Temp. Threshold:     77 Celsius
Critical Comp. Temp. Threshold:     80 Celsius

Supported Power States
St Op     Max   Active     Idle   RL RT WL WT  Ent_Lat  Ex_Lat
 0 +     3.50W       -        -    0  0  0  0        0       0
 1 +     2.70W       -        -    1  1  1  1        0       0
 2 +     2.00W       -        -    2  2  2  2        0       0
 3 -   0.0250W       -        -    3  3  3  3     2000    5000
 4 -   0.0040W       -        -    4  4  4  4     5000    9000

Supported LBA Sizes (NSID 0x1)
Id Fmt  Data  Metadt  Rel_Perf
 0 +     512       0         0

=== START OF SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED
```

ashaduri, 2024-10-23:
Duplicate of #46 and #4.

closed 2024-10-23 by ashaduri