ashaduri/gsmartcontrol #69: Supported Hardware page could be clearer

Chealer, 2024-11-19:
[The _Supported Hardware_ page](https://gsmartcontrol.shaduri.dev/supported-hardware) is very helpful, but I suspect it can be confusing for people with limited experience.

It currently reads:

> GSmartControl supports ATA drives (both PATA and SATA), NVMe drives, various USB to ATA bridges, and drives behind some RAID controllers.
> 
> See [Smartmontools USB Device Support](https://www.smartmontools.org/wiki/Supported_USB-Devices) page for an (incomplete) list of supported USB to ATA bridges.
> 
> GSmartControl supports the following RAID controllers:
> 
> [...]
> 
> Note: Smartmontools supports even [more RAID Controllers](https://www.smartmontools.org/wiki/Supported_RAID-Controllers). The drives behind such controllers can be manually entered in GSmartControl using Add Device… functionality or --add-device command-line option.

I do not know smartmontools well and am far from being sure of what it supports, but if my understanding is correct, I would suggested rephrasing to something like:

> GSmartControl supports ATA drives (both PATA and SATA) and NVMe drives when they are directly attached. For drives behind bridges or RAID controllers, see the following section.
>
> **USB and RAID**
> See [smartmontools USB Device Support](https://www.smartmontools.org/wiki/Supported_USB-Devices) page for an (incomplete) list of supported USB to ATA bridges.
> 
> GSmartControl supports the following RAID controllers:
> 
> [...]
> 
> Note: Smartmontools supports even [more RAID Controllers](https://www.smartmontools.org/wiki/Supported_RAID-Controllers). The drives behind such controllers can be manually entered in GSmartControl using Add Device… functionality or --add-device command-line option.

ashaduri, 2024-11-19:
I updated the [Supported Hardware](https://gsmartcontrol.shaduri.dev/supported-hardware) page for clarity.

Please note that some USB to NVMe bridges are supported as well.

closed 2024-11-19 by ashaduri

Chealer, 2024-11-19:
Thanks @ashaduri 
I wrote this ticket thinking that my own knowledge about hardware has vastly diminished in the last 15 years, and I imagine the text could be intimidating for new generations.

Perhaps it clarifies to specify that SCSI drives are not supported. Also, the new text lists ATA drives, but not which. Perhaps the first 2 items in the list can be numbered and #1 relabelled to "All ATA drives―either serial (SATA) or parallel (PATA)―which support SMART", **if** that is accurate. Then the third element can read "#1 and #2 behind some USB bridges […]".

ashaduri, 2024-11-21:
I added a note about SCSI / SAS support.

I left the rest as is. The page already lists SATA and PATA drives. As for the SMART support, any non-buggy hdd/ssd has SMART support, unless it's connected via USB, and USB bridges are listed separately.
I also do not want to use the term "directly attached" - it is very vague and may be even more confusing.