ashaduri/gsmartcontrol #71: GSmartControl 2.0.1 crashes on Win7

SquirrelACY, 2024-12-16:
**Version and Environment**
 - GSmartControl version: 2.0.1
 - OS: Windows 7 SP1 x64

**Describe the Bug**
When double-clicking on a SATA drive, GSmartControl crashes.

**To Reproduce**
Steps to reproduce the behavior:
1. Start GSmartControl
2. Double-click on a SATA HDD/SSD
3. GSmartControl crashes

**Screenshots**
[image]
[image]

**Additional Context**
This problem does not happen on Windows 8.1 nor Windows 10 22H2 or 21H2 LTSC.
I haven't tried this on Vista. If you're interested to see what happens on this OS, let me know and I'll try and create a lab PC.

Double-clicking SATA optical drives or SCSI memory card readers does not crash the program.

[stderr.txt](https://github.com/user-attachments/files/18157432/stderr.txt)

ashaduri, 2024-12-17:
Hi,
Thanks a lot for the report.

Can you please try the 32-bit version to see if it still crashes?
https://github.com/ashaduri/gsmartcontrol/releases/download/v2.0.1/gsmartcontrol-2.0.1-win32.zip

Also, can you please list the installed drives (are they SATA/SSD, nvme, HDD, etc.)?

Thanks

SquirrelACY, 2024-12-17:
Hi ashaduri,

The 32-bit version also crashes.
Here's the log: [stderr.txt](https://github.com/user-attachments/files/18169708/stderr.txt)

On the system I've provided logs from:
pd0 is an M.2 NMVE drive (under Windows 7 it says "no additional information is available for this drive")
pd1 is a SATA HDD (this is the one that crashes)
/dev/csmi1,1 (ata) is a DVD-RW drive

Thanks

ashaduri, 2025-02-13:
I was able to reproduce this in a VM by clicking the "Refresh" button after opening the Information window.

Found the culprit: One of the dependencies used for icon loading, librsvg 2.59 requires Windows 8.1 or later (confirmed by dependency walker).
Updated the CI to use 2.58 instead, which seems to fix the issue.

The fix will be available in the next release.

closed 2025-02-13 by ashaduri

SquirrelACY, 2025-03-01:
Confirmed this issue is fixed in 2.0.2. Thanks!