ashaduri/gsmartcontrol #77: Missing Installation Instructions and C++20 Compatibility for Pop!_OS 22.04 (LTS)

Photolunatic, 2025-02-06:
**Version and Environment**

     GSmartControl version: Git (building from source)
     OS: Pop!_OS 22.04 (Ubuntu 22.04 LTS derivative)
     Compiler: g++-11 (11.4.0-1ubuntu1~22.04)

**Describe the Bug** 

When attempting to build gsmartcontrol on Pop!_OS 22.04 (LTS) from source, I encounter errors related to missing C++20 features (e.g., std::format). Despite using g++-11, which should support C++20, the build fails with errors like:

`fatal error: format: No such file or directory`

**To Reproduce Steps to reproduce the behavior:**

    Clone the gsmartcontrol repository from GitHub.
    Navigate to the project directory and create a build folder (mkdir build && cd build).
    Run the build command: cmake .. followed by make -j$(nproc).
    See error: Compilation fails due to missing C++20 features (std::format).

**Screenshots** 
[image]

**Additional Context**
I’m using the default system configuration for Pop!_OS 22.04 LTS. This is a widely used LTS version, so I find it odd that there are no packages or repository dedicated for it. I could not find an installation/ building steps in the README. There seems to be no AppImage, or Flatpak available, which is why I tried building from source.

It would be great if the project provided installation steps, along with the necessary prerequisites for systems like 22.04 LTS. Or, if there’s a known workaround for this issue, please let me know.

ashaduri, 2025-02-06:
Hi,
GCC 11 is not enough for proper C++20.

Per GSmartControl's [software requirements](https://gsmartcontrol.shaduri.dev/software-requirements) page, GCC 13 is the minimum supported version.

As for the installation steps, I'll add them to the requirements page.

Photolunatic, 2025-02-06:
Feedback on GCC Requirement and Installation Process

Hi Alex,

Thank you for the response and for planning to update the installation steps on the requirements page.

However, I’d like to express a concern that might affect many users. As you mentioned, GCC 13 is required for proper C++20 support, but many users (especially those on LTS systems like Pop!_OS 22.04, Ubuntu, etc.) are stuck with GCC 11 or 12 as part of their stable repositories. These systems are widely used for their stability, security, and long-term support, but the lack of compatibility with GSmartControl's build process creates a barrier to using your tool.

Moreover, since no official repository, AppImage, or Flatpak is provided, users who cannot or do not want to upgrade GCC to a much newer version are left without a way to use your software. For LTS users, being left with only the option to manually build with potentially breaking dependencies or having to upgrade their system's core components feels like an oversight.

If possible, could you consider adding:

- Official LTS-friendly installation packages (e.g., a PPA for Ubuntu-based distributions or a more broadly compatible AppImage/Flatpak).

- Clarifying in the installation steps that users on LTS versions may face difficulties with GCC version compatibility and offering guidance on safe workarounds.

I’m sure many users like myself would appreciate it, especially considering the popularity of LTS distributions for production environments.

Thanks again for the work you’re doing, and I hope this feedback helps make GSmartControl more accessible to all users.

Best regards,
Seb

ashaduri, 2025-02-07:
It seems you can install gcc 13 on Ubuntu 22.04:
https://linuxcapable.com/how-to-install-gcc-compiler-on-ubuntu-linux/

Having said that, gsmartcontrol 1.1.4 should be available in Ubuntu 22.04 repositories. It works fine if you're not interested in NVMe support.
Since LTS releases are more or less "stuck" on fixed versions apart from a handful of major updates, there should be no expectation of having access to the latest versions of software.

There is a request for AppImage support, see #75. However, it is unlikely that an AppImage built on a modern Ubuntu will work on an older one.

MagicalDrizzle, 2025-02-10:
@Photolunatic As for getting newer gcc - you can probably install it using Homebrew and compile it manually.
And for building AppImage that works on older distros, I believe it depends on the glibc version - so just building it on an older distro should work fine.

ashaduri, 2025-02-13:
@Photolunatic Would you mind trying compiling the git main version?
While I will not be getting rid of the C++20 requirement, I have modified the latest git version to support gcc 11.
It would be great if you could test it.

Thanks

Photolunatic, 2025-02-14:
Hi Alexander,

Thank you for taking the time to address this issue. I truly appreciate your efforts in trying to make this work for users like me.

Unfortunately, I’m a bit time-restrained over the next few days, but I’ll definitely look into testing the new version as soon as I can.

Thanks again for your responsiveness and support!

orhun, 2025-02-22:
Hello, Arch Linux packager here 👋🏼 

I'm trying to package the latest version based on the instructions provided in the requirements page, and ended up with a flow like this:

```
mkdir build
cd build
cmake .. -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Release
make
make DESTDIR="${pkgdir}" install
```

For some reason it installs the binary in `/usr/sbin/` though, it wasn't the case before.
Any idea why that might be happening & how to configure it to install in `/usr/bin` again?

P.S. This is one of my rare cmake packages, so I'm not familiar with all the configuration options.

ashaduri, 2025-02-23:
@orhun 
Try setting `CMAKE_INSTALL_SBINDIR`:
```
cmake .. -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_INSTALL_SBINDIR=/usr/bin -DCMAKE_BUILD_TYPE=Release
```

I just checked and version 1.1.4 (the last one to use autotools) also installs to `/usr/sbin`, because run-as-root binaries go there. Perhaps Arch's default autotools flags overrode that previously.

P.S. Please always open a new issue if it has no direct relation to existing one, even if it's just a question.

Photolunatic, 2025-02-23:
Hi Alexander,

I just tried again, and it builds with not issues.

```
$ cmake .. -DCMAKE_C_COMPILER=gcc-11 -DCMAKE_CXX_COMPILER=g++-11 -G "Unix Makefiles"
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/g++-11 - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using toolchain file: 
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'gtkmm-3.0>=3.0'
--   Found gtkmm-3.0, version 3.24.5
-- Checking for module 'glibmm-2.4'
--   Found glibmm-2.4, version 2.66.2
-- Found Intl: built in to C library  
-- Host system:
-- CMAKE_HOST_SYSTEM_NAME: Linux
-- CMAKE_HOST_SYSTEM_VERSION: 6.9.3-76060903-generic
-- CMAKE_HOST_SYSTEM_PROCESSOR: x86_64
-- Target system:
-- CMAKE_SYSTEM_NAME: Linux
-- CMAKE_SYSTEM_VERSION: 6.9.3-76060903-generic
-- CMAKE_SYSTEM_PROCESSOR: x86_64
-- CMAKE_SIZEOF_VOID_P: 8
-- CMAKE_CXX_COMPILER_ID: GNU
-- CMAKE_CXX_COMPILER_VERSION: 11.4.0
-- CMAKE_CXX_COMPILER: /usr/bin/g++-11
-- CMAKE_CXX_COMPILER_EXTERNAL_TOOLCHAIN: 
-- CMAKE_CXX_COMPILER_TARGET: 
-- CMAKE_CXX_COMPILE_FEATURES: cxx_std_98;cxx_template_template_parameters;cxx_std_11;cxx_alias_templates;cxx_alignas;cxx_alignof;cxx_attributes;cxx_auto_type;cxx_constexpr;cxx_decltype;cxx_decltype_incomplete_return_types;cxx_default_function_template_args;cxx_defaulted_functions;cxx_defaulted_move_initializers;cxx_delegating_constructors;cxx_deleted_functions;cxx_enum_forward_declarations;cxx_explicit_conversions;cxx_extended_friend_declarations;cxx_extern_templates;cxx_final;cxx_func_identifier;cxx_generalized_initializers;cxx_inheriting_constructors;cxx_inline_namespaces;cxx_lambdas;cxx_local_type_template_args;cxx_long_long_type;cxx_noexcept;cxx_nonstatic_member_init;cxx_nullptr;cxx_override;cxx_range_for;cxx_raw_string_literals;cxx_reference_qualified_functions;cxx_right_angle_brackets;cxx_rvalue_references;cxx_sizeof_member;cxx_static_assert;cxx_strong_enums;cxx_thread_local;cxx_trailing_return_types;cxx_unicode_literals;cxx_uniform_initialization;cxx_unrestricted_unions;cxx_user_literals;cxx_variadic_macros;cxx_variadic_templates;cxx_std_14;cxx_aggregate_default_initializers;cxx_attribute_deprecated;cxx_binary_literals;cxx_contextual_conversions;cxx_decltype_auto;cxx_digit_separators;cxx_generic_lambdas;cxx_lambda_init_captures;cxx_relaxed_constexpr;cxx_return_type_deduction;cxx_variable_templates;cxx_std_17;cxx_std_20;cxx_std_23
-- CMAKE_CXX_FLAGS: 
-- CMAKE_CXX_FLAGS_DEBUG: -g
-- CMAKE_CXX_FLAGS_MINSIZEREL: -Os -DNDEBUG
-- CMAKE_CXX_FLAGS_RELEASE: -O3 -DNDEBUG
-- CMAKE_CXX_FLAGS_RELWITHDEBINFO: -O2 -g -DNDEBUG
-- CMAKE_STATIC_LINKER_FLAGS: 
-- CMAKE_STATIC_LINKER_FLAGS_DEBUG: 
-- CMAKE_STATIC_LINKER_FLAGS_MINSIZEREL: 
-- CMAKE_STATIC_LINKER_FLAGS_RELEASE: 
-- CMAKE_STATIC_LINKER_FLAGS_RELWITHDEBINFO: 
-- CMAKE_SHARED_LINKER_FLAGS: 
-- CMAKE_SHARED_LINKER_FLAGS_DEBUG: 
-- CMAKE_SHARED_LINKER_FLAGS_MINSIZEREL: 
-- CMAKE_SHARED_LINKER_FLAGS_RELEASE: 
-- CMAKE_SHARED_LINKER_FLAGS_RELWITHDEBINFO: 
-- CMAKE_EXE_LINKER_FLAGS: 
-- CMAKE_EXE_LINKER_FLAGS_DEBUG: 
-- CMAKE_EXE_LINKER_FLAGS_MINSIZEREL: 
-- CMAKE_EXE_LINKER_FLAGS_RELEASE: 
-- CMAKE_EXE_LINKER_FLAGS_RELWITHDEBINFO: 
-- CMAKE_C_COMPILER: gcc-11
-- Configuring done
-- Generating done
-- Build files have been written to: /home/.../gsmartcontrol/build

```

The configuration step completed successfully.

```
$ make -j$(nproc)
[  4%] Building CXX object dependencies/fmt/CMakeFiles/fmt.dir/fmt/src/format.cc.o
[  4%] Building CXX object dependencies/whereami/CMakeFiles/whereami.dir/whereami/src/whereami.cpp.o
[  6%] Linking CXX static library libwhereami.a
[  6%] Built target whereami
[ 16%] Building CXX object src/libdebug/CMakeFiles/libdebug.dir/dflags.cpp.o
[... 106 lines cut ...]
Critical Warning:                   0x00
Temperature:                        43 Celsius
Available Spare:                    100%
Available Spare Threshold:          10%
Percentage Used:                    0%
Data Units Read:                    23,769,675 [12.1 TB]
Data Units Written:                 21,310,957 [10.9 TB]
Host Read Commands:                 325,545,412
Host Write Commands:                736,255,350
Controller Busy Time:               728
Power Cycles:                       759
Power On Hours:                     4,332
Unsafe Shutdowns:                   20
Media and Data Integrity Errors:    0
Error Information Log Entries:      152
Warning  Comp. Temperature Time:    0
Critical Comp. Temperature Time:    0
Temperature Sensor 1:               36 Celsius
Temperature Sensor 2:               37 Celsius

Error Information (NVMe Log 0x01, 16 of 256 entries)
No Errors Logged
```
Any thoughts where to start debugging this?
Cheers.

ashaduri, 2025-02-24:
Hi @Photolunatic,

Thanks a lot for testing.

Smartctl >= 7.3 is required by GSmartControl 2.0.1.
In theory, it should have supported the older versions of smartctl, but this support was broken.

I fixed it now, so smartctl versions >= 5.43 should be working again (in compatibility mode), though using >= 7.3 is highly recommended.
NVMe is still only supported with 7.3 and newer, and this is not expected to change.

With this, I believe we can close this issue as fixed - GSmartControl will work with gcc11 and smartctl 7.2 starting with 2.0.2.

Thanks

closed 2025-02-24 by ashaduri