ashaduri/gsmartcontrol #78: Wrong temperature number on Windows 10

terremoth, 2025-02-12:
**Version and Environment**
 - GSmartControl version: v2.0.1
 - OS: Windows 10 x64 22H2

**Describe the Bug**
Celsius temperature shows a high bizarre value

[image]

**To Reproduce**
Steps to reproduce the behavior:
1 - enable smart
2 - choose the disk
3 - go to Temperature Log tab
4 - look at the "current temperature"

**Screenshots**

**Additional Context**
maybe this can be useful:

[image]

ashaduri, 2025-02-12:
Hi,
Thanks for the report.

Can you please save the output (in the Information window, "Save As" button), and attach it here?

terremoth, 2025-02-12:
Yes, here it goes:

[Hitachi_HTS547575A9E384_J2140059DJZLSA_2025-02-12_1125.json](https://github.com/user-attachments/files/18769274/Hitachi_HTS547575A9E384_J2140059DJZLSA_2025-02-12_1125.json)

ashaduri, 2025-02-12:
Thanks for the file.
I can see why this happens, and the issue should be fixed in the next release.

terremoth, 2025-02-12:
I checked other HDD with the same problem, this is the 3rd one, maybe  it is happening with all of them on Windows (at least):

[SAMSUNG_HM321HI_S2K5J56B324420_2025-02-12_1147.json](https://github.com/user-attachments/files/18769545/SAMSUNG_HM321HI_S2K5J56B324420_2025-02-12_1147.json)

ashaduri, 2025-02-14:
Fixed in git main branch.

closed 2025-02-14 by ashaduri