ashaduri/gsmartcontrol #79: Self test failed "check log" but no log has been generated

terremoth, 2025-02-12:
**Version and Environment**
 - GSmartControl version: v2.0.1
 - OS: Windows 10 x64 22H2

**Describe the Bug**
No log has been generated after execute self test, says some error occurred but no logs 🤔

**To Reproduce**
Steps to reproduce the behavior:
1. Select the disk with smart enabled
2. Go to Self Tests tab
3. Click on Execute
4. After a minute or less, a message says "TEST FAILED!" in red, "Teste result: Completed With Errors." - Check the Self-Test Log for more information, but no log generates so I don't know what problem my HDD has.

**Screenshots**

[image]

**Additional Context**

Maybe this log can give some useful info:

[SAMSUNG_HM251JI_S1KEJ56S725035_2025-02-12_1103.json](https://github.com/user-attachments/files/18769220/SAMSUNG_HM251JI_S1KEJ56S725035_2025-02-12_1103.json)

terremoth, 2025-02-12:
Ah, this HDD also suffers from the same problem from #78

ashaduri, 2025-02-12:
Thanks for the report and the file.
I will investigate.

P.S. Unless you're already aware of this, please keep in mind that the drive has uncorrected sectors and may die soon.

terremoth, 2025-02-12:
> Thanks for the report and the file. I will investigate.
> 
> P.S. Unless you're already aware of this, please keep in mind that the drive has uncorrected sectors and may die soon.

Yeah, thanks, this drive may die soon indeed. I will put only non disposable data in it. It also show some "pre-failure" items. But I wish I could know more about.

ashaduri, 2025-02-14:
The drive supports only standard self-test log and not extended log.
The new JSON parser did not account for the standard log.

This has now been fixed in git main.

closed 2025-02-14 by ashaduri