ashaduri/gsmartcontrol #80: GSmartControl 1.1.4+ crashes on Vista

SquirrelACY, 2025-03-01:
**Version and Environment**
 - GSmartControl version: 1.1.4 and newer (up to 2.0.2)
 - OS: Windows Vista SP2 x64

**Describe the Bug**
Upon launching GSmartControl, the program crashes.

**To Reproduce**
Steps to reproduce the behavior:
1. Run gsmartcontrol.exe as an admin.
2. GSmartControl crashes.

**Screenshots**
GSmartControl 1.1.4:
[image]

GSmartControl 2.0.2:
[image]

**Additional Context**
I have tried both the 32 and 64 bit versions of GSmartControl. Both behave identically.
GSmartControl 1.1.3 is the newest version I was able to use on Vista.
Running GSmartControl with the -v flag does not produce any log files.

closed 2025-03-02 by ashaduri

SquirrelACY, 2025-03-03:
I agree it's not reasonable to support Vista anymore. But why update the website to say that Vista is limited to version 0.9.0 instead of version 1.1.3?

ashaduri, 2025-03-03:
Versions 1.1.3 has a critical issue (crash) that could be present in all pre-1.1.4 versions, but not in 0.9.x.