ashaduri/gsmartcontrol #81: Incorrect presentation of the SMART attribute 231

glorifyday, 2025-03-05:
**Version and Environment**
 - GSmartControl version: 2.0.2 (also on 1.1.3)
 - OS: Windows 10 Pro 22H2 OS build 19045.5487 (also on Raspbian GNU/Linux 11 (bullseye), Linux 6.1.21-v8+, arm64, respectively)
 - drivedb.h 7.3/5660 (up to date)
 
**Describe the Bug**
Dor some SSDs, the S.M.A.R.T. attribute 231 is presented in GSmartControl (both Raspbian and Windows version) as "Temperature" and false warnings are shown.
On the other hand, the smartctl output says: 231 Unknown_SSD_Attribute.

This attribute might be vendor specific, but for instance Kingston lists it as "SSD Life Left".
https://media.kingston.com/support/downloads/mkp_306_smart_attribute.pdf

I've got four SSDs that use this attribute:
- Kingston XS1000 2 TB,
- Kingston DataTraveler Max 256,0 GB
- Transcend ESD310C 256,0 GB
- GoodRam IR-SSDPR-S25A-240 240,0 GB

For the first three, GSmartControl presents this attribute as "Temperature" (unlike smartctl) and shows a (false) warning. 
For the last one, the presentation is the same as for smartctl and no warning is shown.

**To Reproduce**
Steps to reproduce the behavior:
1. Vie the device details.
2. Go to 'Attributes' tab.
3. Click on 'View Output'
4. Compare the names of the attribute 231.

Incorrect presentation (Kingston, Transcend):
[image]

Correct presentation (GoodRam):
[image]

**Additional Context**
The problem seems to occur in different platforms and software versions.

Included please find the zipped screenshots and JSON outputs from the Windows 2.0.2 version for all the mentioned drives.

[Issue.zip](https://github.com/user-attachments/files/19080814/Issue.zip)

checktext00, 2025-04-16:
It's because smartmontools (`smartctl`) returns `Unknown_SSD_Attribute` if the drive is an SSD and there is no SSD attribute for it under (either the DEFAULT model, or under a specific drive model) in the `drivedb.h` database.

What's happening is that in the `drivedb.h` database there is a [DEFAULT model attribute #231 for _HDDs only_](https://github.com/smartmontools/smartmontools/blob/0372a234d7677a032841e31f267ca8e7e5dc559b/smartmontools/drivedb.h#L143):
```
    "-v 231,raw48,Temperature_Celsius,HDD "
```
and GSmartControl seems to be reading that attribute name, not realizing the drive is an SSD

----

the `smartctl` display logic is in [atacmds.cpp](https://github.com/smartmontools/smartmontools/blob/0372a234d7677a032841e31f267ca8e7e5dc559b/smartmontools/atacmds.cpp#L2137):

```
// Get attribute name
std::string ata_get_smart_attr_name(unsigned char id, const ata_vendor_attr_defs & defs,
                                    int rpm /* = 0 */)
{
  if (!defs[id].name.empty())
    return defs[id].name;
  else {
     const ata_vendor_attr_defs::entry & def = get_default_attr_defs()[id];
     if (def.name.empty())
       return "Unknown_Attribute";
     else if ((def.flags & ATTRFLAG_HDD_ONLY) && rpm == 1)
       return "Unknown_SSD_Attribute";
     else if ((def.flags & ATTRFLAG_SSD_ONLY) && rpm > 1)
       return "Unknown_HDD_Attribute";
     else
       return def.name;
  }
}
```

checktext00, 2025-04-16:
If you think the attribute should have a specific name (like "SSD Life Left") instead of "Unknown_SSD_Attribute", you can update the `drivedb.h` database.

First download the latest version of the `drivedb.h` database using the smartmontools `update-smart-drivedb` command (see `man update-smart-drivedb`), or GSmartControl > Options > Update Drive Database, which should download the latest database from (as of today) https://svn.code.sf.net/p/smartmontools/code/branches/RELEASE_7_3_DRIVEDB/smartmontools/drivedb.h

Then when you run `smartctl` (or view the output from GSmartControl), check that the database is updated with the line:
```
Device is:        In smartctl database 7.3/5671
```
(check that it matches the latest database number on the web from the [drivedb.h](https://svn.code.sf.net/p/smartmontools/code/branches/RELEASE_7_3_DRIVEDB/smartmontools/drivedb.h) link above)

If the attribute still shows "Unknown_SSD_Attribute", you can [create an issue with smartmontools](https://github.com/smartmontools/smartmontools/issues) to update the `drivedb.h` database