ashaduri/gsmartcontrol #84: When using the Nbase DS-80NU3B Docking, the program does not recognize NVMe drives

rozsazoli, 2025-03-18:
Version and Environment
 - GSmartControl version: 2.0.2
 - OS: Windows 11 24H2 build 261003476

Describe the Bug
- When using the Nbase DS-80NU3B Docking, the program does not recognize NVMe drives.

Additional Context 
- Több NVMe kártyával teszteltem, és mindegyiken megtalálható a hiba.

[image]

rozsazoli, 2025-03-18:
[image]

rozsazoli, 2025-03-18:
solution: [here](https://github.com/smartmontools/smartmontools/issues/331)

ashaduri, 2025-03-18:
@rozsazoli You can set the manual smartctl options in Preferences.