ashaduri/gsmartcontrol #87: Need way to add more devices, Shortcut can't accept long enough string

LostOnTheLine, 2025-04-28:
**Version and Environment**
 - GSmartControl version: 2.0.2
 - OS: Windows 11

**Is your feature request related to a problem? Please describe.**
I have 2 4-bay RAID controllers for a total of 8 disks that need to be added.  
In the `[Add Device]` window it says `Note: to make this change permanent, you'll have to add the gsmartcontrol --add-device command line option to it's shortcut.` but there is only enough space in the shortcut `Target:` field to put `"C:\Program Files\gsmartcontrol\gsmartcontrol.exe" --add-device /dev/sdd::jmb39x-q,0 --add-device /dev/sdd::jmb39x-q,1 --add-device /dev/sdd::jmb39x-q,2 --add-device /dev/sdd::jmb39x-q,3 --add-device /dev/sdc::jmb39x,0 --add-device /dev/sdc::jmb39x,1` but I still need to add 2 more devices ` --add-device /dev/sdc::jmb39x,2 --add-device /dev/sdc::jmb39x,3`. It would be nice if we could add them to a config file. I tried adding them with flags under the `[Preferences]` > `{Drives}` but it doesn't stick unless you add `Perameters:` but even if they are added they do not add the drives. But if I try to run it as a batch file it can't find the smartctl binary & it forces a terminal window open while it's running. 

**Describe the solution you'd like**
Can we have an option to run a config file or even use a file instead of all the `--add-device` options like
```powershell
"C:\Program Files\gsmartcontrol\gsmartcontrol.exe" --config C:\tools\smartmontools\drives.conf
or 
"C:\Program Files\gsmartcontrol\gsmartcontrol.exe" --config C:\tools\smartmontools\drives.yaml
```
or even just a single-line text file with all the flags
```powershell
"C:\Program Files\gsmartcontrol\gsmartcontrol.exe" --config C:\tools\smartmontools\drives.txt
```
that could contain
```txt
--add-device /dev/sdd::jmb39x-q,0 --add-device /dev/sdd::jmb39x-q,1 --add-device /dev/sdd::jmb39x-q,2 --add-device /dev/sdd::jmb39x-q,3 --add-device /dev/sdc::jmb39x,0 --add-device /dev/sdc::jmb39x,1 --add-device /dev/sdc::jmb39x,2 --add-device /dev/sdc::jmb39x,3
```

**Describe alternatives you've considered**
As it is now it's unusable for anyone with more than a small handful of drives that need custom device calls, & anyone with a RAID that has more than 6 bays or multiple RAID arrays the only option is to use a different program, but there aren't any 

**Additional context**
Add any other context or screenshots about the feature request here.

ashaduri, 2025-04-28:
Hi,

So, this is essentially a limit of the Windows shortcut properties dialog?

I agree that having a persistent list of added devices is a proper solution to this, but I'm not sure when this feature will be added.

Meanwhile, have you tried running your batch file as Administrator (perhaps creating a shortcut to .bat and checking the "Run as Administrator" checkbox in its properties)? Also, you can specify the full path to smartctl-nc.exe in Preferences in case the batch file breaks the file search somehow.

closed 2025-04-28 by ashaduri

LostOnTheLine, 2025-05-06:
> This has now been fixed - manually added devices persist across restarts unless removed manually, or via `--forget-devices`.

Thanks a bunch.

> Meanwhile, have you tried running your batch file as Administrator (perhaps creating a shortcut to .bat and checking the "Run as Administrator" checkbox in its properties)? Also, you can specify the full path to smartctl-nc.exe in Preferences in case the batch file breaks the file search somehow.

I had actually tried more or less all these things, I had tried using the path in the `Start in:` for the shortcut, & just using the filename, but the path was still too long. Running as a `.bat` no matter what I did ran with a terminal window open. I believe I did try running as Admin & had the same issues.

But thank you for the update but I do not see a new version of the Binary anywhere

ashaduri, 2025-05-06:
> I do not see a new version of the Binary anywhere

This feature will be available in the next version when it's released (no definite date yet).